pavel-demin/red-pitaya-notes #1097: Red Pitaya Image

zzyzzy42, 2023-06-18:
### Description of the setup:

 * Device: Red Pitaya 125-14 Z7020
 * SD card image:  red-pitaya-alpine-3.14-armv7-20220322 and red-pitaya-alpine-3.14-armv7-20211119
 * Application: Getting Red Pitaya to work with GNU Radio
 * Other relevant information: Windows 11

### Description of the problem:
Hi, I tried both of the listed "images" and couldn't get Red Pitaya to work. I feel like I am missing something. I followed the instructions to download the "images" for Red Pitaya, but those "images" are zip files without any .img file inside. The size of the files is also very small compared to the factory image.
I am a newbie. Sorry if this question sounds stupid. Thank you soooo much for your effort!

### Steps to reproduce the problem:

 1. Format my SD card.
 2. Download the Red Pitaya images under _Getting started with GNU Radio_.
 3. Unzip the images and ctrl+c and ctrl+v all the files inside to the SD card.
 4. Plug the SD card into my Red Pitaya 125-14 Z7020.
 5. Connect an ethernet cable to my laptop running Windows 11 with GNU Radio.
 6. Power on the Red Pitaya.
 7. The Red Pitaya only has a green LED lit and also a yellow LED dimly on.
 8. Cannot connect to the Red Pitaya using 192.168.1.100.
 9. I also tried to connect the Red Pitaya wirelessly by connecting it to my router, but I don't see it on my router admin page.

pavel-demin, 2023-06-18:
Have you tried the troubleshooting instructions at the following link?

http://pavel-demin.github.io/red-pitaya-notes/alpine/#troubleshooting

the Ethernet interface of the computer should be configured to have an IP address in the same 192.168.1.x sub-network. For example, 192.168.1.111. Instructions on how to set a static IP address in Windows can be found at [this link](https://kb.netgear.com/27476/How-do-I-set-a-static-IP-address-in-Windows).
The pre-built apps on the SD card are for the original 125-14 Z7010 board, not the 125-14 Z7020 board. So you will have to build the bitstream files yourself.

zzyzzy42, 2023-06-18:
Hi Pavel,
Thank you so much for your speedy reply!
I found another Red Pitaya STEM122-16 in my lab. I tried both images listed in my original post, and they did not work either.
I was expecting this result because you mentioned that the pre-built image would not work with Z7020 chips. Does this sound right?

Thanks a lot for your help!

pavel-demin, 2023-06-19:
This SD card image contains pre-built applications for the 125-14 Z7010 and 122-16 Z7020 boards. This SD card image can also be used to boot the 125-14 Z7020 board but the applications on the SD card are not compatible with the 125-14 Z7020 board.

It is normal that there are no blinking LEDs after booting the Red Pitaya board with this SD card image.

Please try to follow the [troubleshooting instructions](http://pavel-demin.github.io/red-pitaya-notes/alpine/#troubleshooting).

The first step is to check the boot process using the USB/serial console as explained at [this link](https://redpitaya.readthedocs.io/en/latest/developerGuide/software/console/console/console.html).

If the boot process does not work at all, then there is a problem with the SD card.

closed 2023-06-21 by pavel-demin

zzyzzy42, 2023-06-24:
Hi Pavel,

Thanks for your help! I will try that!
Have a great weekend!