pavel-demin/red-pitaya-notes #1099: How about using SpinalHDL to create Red-Pitaya firmware?

volatile-static, 2023-06-28:
Hello, I am a user of Red Pitaya and SpinalHDL.  This repository shows the powerful features of Red Pitaya, and I sincerely appreciate your work.
I noticed that you are using SpinalHDL recently, and I think this is a very smart choice, because SpinalHDL is more concise, powerful, and flexible than VHDL or Verilog. I would like to ask you if you are willing to add SpinalHDL as an option to the red-pitaya-notes repository. I am considering using SpinalHDL to implement an MRI system and experiment on red-pitaya, I hope to get your guidance.
Thank you for your contribution to the open-source community, and I wish you all the best.

pavel-demin, 2023-06-28:
I would not say that I really use SpinalHDL. I only played a little with Chisel and SpinalHDL last year. I also looked at Clash, Hardcaml, Amaranth, Magma and a few other HDL languages and libraries.

These languages are probably useful for writing highly reconfigurable CPUs but I have not found how to use them effectively for what I am doing and what I would be interested in doing.

Using these languages to only write IP cores would be a waste of their potential. They should also be used to instantiate and interconnect IP cores. This means that the combination of Verilog and Tcl that I am currently using should be replaced with one of these languages.

Most Verilog modules in this repository are either a few lines of code or blackbox (DSP48E1, XADC, xpm_fifo_sync) wrappers. Tcl is still required to make use of the DSP IP cores (DDS, CIC, FIR) that come with Vivado.

Moving to Chisel or SpinalHDL or Magma or ... would require rewriting the DSP IP cores and that is something I would be interested in doing one day. However, it is too complicated and time-consuming for me right now.

Another problem with these languages is that there are so many of them and it is not easy to choose and their future is unclear. Currently, I think that Chisel seems like a better choice because some big companies are interested in it and the community around it offers books and online lectures.

volatile-static, 2023-06-28:
Thank you for your reply. 
My opinion is that these languages are indeed overkill for just writing IP cores, and Tcl might be a better option for these projects, but I have not learned Tcl, so I prefer to use the way I am familiar with to develop. As for the choice between Chisel and SpinalHDL, this is a long-standing issue, and I thank you for sharing your opinion, but we are not suitable to discuss it here.

closed 2023-06-28 by volatile-static