pavel-demin/red-pitaya-notes #1100: error when sourcing the start.sh

Jraviteer, 2023-06-30:
### Description of the setup:

 * Device: 125-14-z7020-LN
 * SD card image: red-pitaya-alpine-3.14-armv7-20220322
 * Application: pulsed-nmr
 * Other relevant information:

### Description of the problem:

### Steps to reproduce the problem:

 1. ssh to redpitaya using powershell
 2. followe the code in the scressnshots

Hello, I was wondering if you could help me how to solve this error I'm getting?
[image]

This is the connection:
[image]

pavel-demin, 2023-06-30:
I normally run `start.sh` scripts and do not source them, but I think sourcing them should also be OK.

This SD card image contains pre-built applications for the 125-14 Z7010 and 122-16 Z7020 boards. The applications on the SD card are not compatible with the 125-14 Z7020 board.

To run the pulsed NMR application on the 125-14 Z7020 board, the `pulsed_nmr.bit` file should be rebuilt using the following command:
```
make NAME=pulsed_nmr PART=xc7z020clg400-1 bit
```

The 125-14 Z7020 board seems to be a fairly recent model. I have never had one or tested my applications with it.

closed 2023-07-01 by pavel-demin

Jraviteer, 2023-07-02:
Ah, I see. Thank you. I was wondering how you generated these Makefile and pulsed-nmr file in the image below? So, I can make my one own for the z7020!
[image]

pavel-demin, 2023-07-03:
Normally, it should be enough to rebuild `pulsed_nmr.bit` using the source code corresponding to the version of the SD card image you are using:
https://github.com/pavel-demin/red-pitaya-notes/releases/tag/20220322

There is no need to rebuild `pulsed-nmr`. It is compatible with Z7010 and Z7020 as they have the same CPU and only the FPGAs are different.

If you want to update `pulsed-nmr.c` to the latest version and rebuild `pulsed-nmr`, then the following commands can be used:
```
apk add gcc make
cd apps/pulsed_nmr
rw
curl -LO https://raw.githubusercontent.com/pavel-demin/red-pitaya-notes/master/projects/pulsed_nmr/server/pulsed-nmr.c
make clean
make
ro 
```

In this case, you will also need to rebuild `pulsed_nmr.bit` from the latest version of the source code using the following commands:
```
git clone https://github.com/pavel-demin/red-pitaya-notes
cd red-pitaya-notes
make NAME=pulsed_nmr bit
```

Jraviteer, 2023-07-05:
Thank you!
So, I have changed the c file and the bit file as you have advised, however, whenever I run this code: "sh start.sh"
It is still giving this error and I have to reboot the red pitaya to be able to connect again. I don't really understand what 'mv10' is here.
[image]

pavel-demin, 2023-07-05:
The 'mvl0' message is not related to this issue.

It still seems that you are using `pulsed_nmr.bit` incompatible with your board.

What happens is that when the `start.sh` script uses incompatible `pulsed_nmr.bit` to configure the FPGA, the FPGA is not configured correctly. When the `pulsed-nmr` program tries to communicate with the unconfigured FPGA, the system freezes.

Jraviteer, 2023-07-05:
I changed it to z7020 through vivado. Does that mean, there are additional things I have to do to make it work?

pavel-demin, 2023-07-05:
I can build a custom SD card image for you with just the `pulsed_nmr` project and with the settings that I think should work. If you are OK with testing it, then I can send you a link later today or tomorrow.

Jraviteer, 2023-07-05:
That would be amazing. I'm happy to test it, thank you :) Please note that I'm located in Australia and so my responses will be very irregular. Apologies for the inconvenience.

pavel-demin, 2023-07-05:
Here is a link to an SD card image with the `pulsed_nmr` project built for `xc7z020clg400-1`:

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADjAc4-tZHnmz5A13EudYO_a/pulsed_nmr/red-pitaya-alpine-3.18-armv7-20230705-pulsed_nmr-z7020.zip?dl=1

It starts on the STEMlab 122.88-16 board that has the same FPGA without freezing the system.

The `start.sh` script runs automatically at boot time.

Jraviteer, 2023-07-06:
Thank you! Can I ask what the password is? I typed RedPitaya but it didn't work. :(

Jraviteer, 2023-07-06:
So I have booted it into mine, and it hasn't kicked me out yet. So, I'm gonna assume that it's working. To run the c file, I have been running it on spyder but would you recommend running it on the alpine instead? And if so, can I ask what code  to insert? Because the tutorial I followed from redpitaya main page did not work.

pavel-demin, 2023-07-06:
Thank you for the test.

I am afraid that I do not understand the questions from your last comment.

If they are about the control program (`pulsed_nmr.py`), then it requires Python3, Matplotlib and PyQt5 or PySide2.

It is also possible to run the control program using the same Python environment used for the MCPHA and VNA projects. Download and unzip the [release zip file](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20230705/red-pitaya-notes-20230705.zip). Copy `pulsed_nmr.py` and `pulsed_nmr.ui` to the `control` directory, then copy `vna.exe` to `pulsed_nmr.exe`.

Only `pulsed_nmr.py` does not do much. There is still a lot to do to make it useful.

Jraviteer, 2023-07-07:
Ah yes, the control program. I apologize for confusing you. So, when I run the code from the client folder, this is what happens. I don't really know what I am doing wrong. The redpitaya is also connected to the computer.

[image]

pavel-demin, 2023-07-07:
I would say that the IP address is not correct. It is also possible to use the mDNS hostname (`rp-f0ae0b.local`) in place of the IP address.

Jraviteer, 2023-07-09:
[image]
Still no luck... Do you have any idea what could cause this?
[image]

pavel-demin, 2023-07-10:
I do not understand the meaning of the first screenshot.

To me it still looks like a network connection issue between the computer you are running the client program on and the Red Pitaya board. Since I do not know anything about your network, I have no idea what it might be.

Can you connect from this computer to the board via SSH?

Jraviteer, 2023-07-10:
Sorry, the first screenshot is just the IP address of the Red pitaya printed out with code 'arp -g' from powershell. The computer is connected to the wifi through a dongle and same with red pitayta. Lastly yes, I'm able to connect it via ssh. :((

pavel-demin, 2023-07-10:
Have you tried using with the client program the same IP address that works for SSH?

Otherwise, I am afraid I cannot help you with this problem.

Jraviteer, 2023-07-19:
Sorry, for the late reply! Now, it works fine. I have no clue, where the error came from but now the error is gone. Can I ask what you have changed in the setting for it to work?

pavel-demin, 2023-07-20:
I am glad that it finally works for you. I did not change anything.

SoukeyCircuits, 2025-11-18:
Hello, sorry for bringing this back up but I was wondering if the most up to date version of the software has fixed the compatibility issue for **pulsed-nmr** with the **125-14-z7020-LN** device when running the image **red-pitaya-alpine-3.22-armv7-20251012**.

I've got both a TRX_Duo and the 7020 version red pitaya that I've been trying to get a pulse out of and I haven't been successful with running the visual studio python applications on either. The TRX is at least able to connect to the board whereas the 7020 can't making me think this chain might have my solution.

-Anthony

pavel-demin, 2025-11-18:
> fixed the compatibility issue for pulsed-nmr with the 125-14-z7020-LN device

The compatibility issue is on your end, as you are using hardware that my projects were not designed to work with. So it is up to you to fix this compatibility issue.

I have already suggested a workaround and you have confirmed that it helped you. Please do not expect anything more from me than I have already done.

All the projects in this repository are designed for the Red Pitaya boards listed on the main page of my notes: initially Red Pitaya STEMlab 125-14, and later Red Pitaya SDRlab 122-16.

These are my personal hobby projects. When I started working on them in 2014-2015, I opted for the original Red Pitaya board because of its very affordable price and wide availability.

I do not own the boards that you mentioned in your previous comment, I have no intention of buying them due to their limited availability and excessive price, and I have no intention of dedicating my hobby time to supporting the entire range of boards that Red Pitaya has decided to produce in recent years.