pavel-demin/red-pitaya-notes #1101: Question about cores

svhum, 2023-07-06:
Hi, I noticed a lot of code refactorization happening recently, and was simply curious as to why the cores no longer include `core_config.tcl` files? Previously cores had parameter descriptions and interface modes fixed by the script, which seems helpful in keeping the cores easier to use (or harder to misuse). Now it seems much of the interfaces are inferred by Vivado and parameter documentation removed. Why the change?

pavel-demin, 2023-07-07:
Thank you for your feedback.

I did not realize that the `core_config.tcl` scripts were so useful. For me, they were just an annoying necessity to be able to use some Verilog modules with the IP cores provided by Xilinx.

I have been curious to try the RTL modules in IP integrator Vivado for quite some time. I finally tried them this year, but unfortunately they do not work well. Apparently this is a known issue that is still unresolved:
https://support.xilinx.com/s/question/0D52E00006hpeJRSAY/vivado-20172-srcscanner-delays-synthesis-for-a-long-time

While trying the RTL modules, I also realized that the `core_config.tcl` scripts are not really necessary anymore. So I was happy to be able to get rid of them.

svhum, 2023-07-07:
Thanks for your response. I didn't realize there were these issues with the IP integrator, so shedding the files makes sense if it's not practical to use the cores there. I also didn't realize previously that Vivado does a fairly good job at inferring the bus interfaces from signal names, making some of what the `core_config.tcl` file redundant.

closed 2024-02-17 by pavel-demin