pavel-demin/red-pitaya-notes #1102: Connecting a Red Pitaya and Raspberry Pi Using an Ethernet Cable

zzyzzy42, 2023-07-13:
### Description of the setup:

 * Device: STEM122-16 and Raspberry Pi 4B
 * SD card image: red-pitaya-alpine-3.14-armv7-20220322
 * Application: GNU Radio
 * Other relevant information:

### Description of the problem:
Hi,
I am trying to connect my Red Pitaya to my RPi using the wired method but couldn't get it to work. They communicate wirelessly via a router with no problem. Are there idiot-proof instructions on how to get Red Pitayas to work with a Raspberry Pi using an ethernet cable?

I'm suspecting this is related to a different subnetwork issue (RPI eth1 address: 169.254.10.103/16), but I can't find any helpful resources on how to fix this on a Linux device. I have tried a few methods and ended up breaking my wifi function of RPI lol.

Thank you soooo much for your help!

### Steps to reproduce the problem:

 1. Connect my Red Pitaya to my RPi using an ethernet cable and ethernet-to-USB adaptor
 2. Enter rp-fxxxxx.local, 169.254.10.103/16, or 192.168.1.100 to Chromium
 3. Get "This site can't be reached" or the device refuses to connect

pavel-demin, 2023-07-13:
I think the easiest would be to connect both Red Pitaya and Raspebery Pi to a router using an Ethernet cable. This way the router will assign correct IP addresses to both devices.

If you really need to connect the two boards with a single Ethernet cable, then the Ethernet interface of the Raspberry Pi board should be configured to have an IP address in the same 192.168.1.x sub-network. For example, 192.168.1.111. According to how-to guides that I found on the web, this can be done with the following commands:

- open the configuration file with:
```
sudo nano /etc/dhcpcd.conf
```
- paste these lines at the end of the file:
```
interface eth0
static ip_address=192.168.1.111/24
static routers=192.168.1.1
static domain_name_servers=192.168.1.1
```

zzyzzy42, 2023-07-14:
Hi Pavel,

That worked! Thank you soooooooo much!

May I please ask how you got the working ip_address (the /24 thing), routers (192.168.1.1), and domain_name_servers (192.168.1.1)? I tried to assign them a few random addresses and ended up bricking my RPi every time lol.

Sorry if the question sounds dumb. I'm a noob. Thanks!

pavel-demin, 2023-07-14:
I am glad that I could help you solve this problem.

The network configuration is simply copied from the following link:

https://raspberrytips.com/set-static-ip-address-raspberry-pi/

closed 2023-07-14 by pavel-demin

zzyzzy42, 2023-07-15:
Hi Pavel,

Thank you so much for all your help! Have an awesome weekend!