pavel-demin/red-pitaya-notes #1104: All my new SD cards boots, but stays forever on "ssh-keygen: generating new host keys: RSA"

Fihl, 2023-07-21:
### Description of the setup:

 * Device: v1r1, I guess, the first Red Pitaya versions
 * SD card image: ..../releases/download/20230705/red-pitaya-alpine-3.18-armv7-20230705.zip
 * Application: sdr-receiver-hpsdr-122-88, but not getting so far yet
 * Other relevant information: Boots correct on older WSPR that has run for years unattended. I try to make a new, for RBN CW-Skimmer

### Description of the problem:
The boot process looks correct, but the line 
  ssh-keygen: generating new host keys: RSA
takes forever, and still there. 

I have tried different SD cards, same result

Installing packages to root filesystem...   done ok, (of 112 items)
 OpenRC 0.47.1 is starting up Linux 6.1.32-xilinx (armv7l)

### Steps to reproduce the problem:

 1. Boot on fresh SD card, and wait, output on PuTTY
 2.
 3.

[booting.txt](https://github.com/pavel-demin/red-pitaya-notes/files/12131366/booting.txt)

Fihl, 2023-07-22:
Forgot to tell that I also copied start.sh from apps/sdr_receiver_hpsdr_122_88 to root folder. 
Removing this, I now gets to login
(I did as explained, but optional from https://pavel-demin.github.io/red-pitaya-notes/sdr-receiver-hpsdr-122-88/ )

Sorry for confusion
Christen, OZ1AAB

[image]

pavel-demin, 2023-07-22:
I tried to reproduce the problem with and without the `start.sh` script. In both cases, the key generation takes some time but does not hang.

Since your system boots now, you can run the following command to save the generated keys and they will not be regenerated on the next boot:
```
lbu commit -d
```

Fihl, 2023-07-22:
Thanks Pavel

I also did mess around in  versions, as my Pitaya being old, I should not try starting sdr-receiver-hpsdr-122-88, 
but instead I should use the basic one sdr-receiver-hpsdr, without the 122/125 suffix

CwSkimmer has located it now, looking to be a nice weekend

Thanks

closed 2023-07-22 by Fihl