pavel-demin/red-pitaya-notes #1105: Multiple Red Pitayas on a Single Device Using Ethernet Cables

zzyzzy42, 2023-07-22:
### Description of the setup:

 * Device: STEM122-16 and Raspberry Pi 4B
 * SD card image: red-pitaya-alpine-3.14-armv7-20220322
 * Application: GNU Radio
 * Other relevant information:

### Description of the problem:
Hi Pavel,

I hope your week is going well! Sorry to bother you again.

I am wondering if there is a way to connect multiple Red Pitayas to a single device using ethernet cables.

I connected 2 Red Pitayas to my Raspberry Pi using 2 Ethernet cables and tried to access them using their mDNS IP address, the RP-F0XXX.LOCAL/ thing. It seems like I could only connect to one of them.

I must be missing something. Maybe there is a way to set the IP address of different Red Pitayas in their configuration files? Thanks a lot for your help!

### Steps to reproduce the problem:

 1. Connect 2 Red Pitayas to a Raspberry Pi 4B using 2 Ethernet cables.
 2. Set the IP address of the two Ethernet ports on the Raspberry Pi 4B to 192.168.1.111 and 192.168.1.112.
 3. Power on the Red Pitayas.
 4. Open a browser and enter RP-F0899D.LOCAL/. Red Pitaya #1 can now work with GNU Radio.
 5. Open a new browser tab and enter RP-F08940.LOCAL/. It still connects to Red Pitaya #1, even though the IP address is from the other Red Pitaya (Red Pitaya #2).
 6. The same phenomenon is observed in GNU Radio.

pavel-demin, 2023-07-22:
The easiest solution is to use a small router with several gigabit ports and connect all devices to this router.

Here are some examples of such routers:
https://www.tp-link.com/us/business-networking/omada-sdn-router/er605
https://www.zyxel.com/global/en/products/wifi-router/ac750-dual-band-wireless-gigabit-router-nbg6515

If you really insist on your current network setup, then check out the [Network configuration](http://pavel-demin.github.io/red-pitaya-notes/alpine/#network-configuration) notes. The commands to edit [/etc/dhcpcd.conf](https://github.com/pavel-demin/red-pitaya-notes/blob/master/alpine/etc/dhcpcd.conf) are
```
nano /etc/dhcpcd.conf
lbu commit -d
```

zzyzzy42, 2023-07-23:
Will try, thank you very much!

closed 2023-07-23 by zzyzzy42