pavel-demin/red-pitaya-notes #1106: Error when 'source .../block_design.tcl'

marcomeixner, 2023-08-09:
Hallo,

### Description of the setup:

 * Device: RP-STEM 125-10
 * SD card image: STEMlab_125-xx_OS_1.04-18_stable.img.zip
 * Application: pulsed_nmr_122_88
 * Other relevant information:

### Description of the problem:

Vivado is running smoothly for the LED blinker project - so that should be fine. I can open the files and make the Red Pitaya blink.

However, when I want to open the block_design.tcl file the following happens:

source C:/redPitaya/pulsed_nmr_122_88/block_design.tcl
# cell xilinx.com:ip:clk_wiz pll_0 {
#   PRIMITIVE PLL
#   PRIM_IN_FREQ.VALUE_SRC USER
#   PRIM_IN_FREQ 122.88
#   PRIM_SOURCE Differential_clock_capable_pin
#   CLKOUT1_USED true
#   CLKOUT1_REQUESTED_OUT_FREQ 122.88
#   CLKOUT2_USED true
#   CLKOUT2_REQUESTED_OUT_FREQ 245.76
#   CLKOUT2_REQUESTED_PHASE 157.5
#   CLKOUT3_USED true
#   CLKOUT3_REQUESTED_OUT_FREQ 245.76
#   CLKOUT3_REQUESTED_PHASE 202.5
#   USE_RESET false
# } {
#   clk_in1_p adc_clk_p_i
#   clk_in1_n adc_clk_n_i
# }
invalid command name "cell"
    while executing
"cell xilinx.com:ip:clk_wiz pll_0 {
  PRIMITIVE PLL
  PRIM_IN_FREQ.VALUE_SRC USER
  PRIM_IN_FREQ 122.88
  PRIM_SOURCE Differential_clock_capable_pin
  ..."
    (file "C:/redPitaya/pulsed_nmr_122_88/block_design.tcl" line 2)

Thanks a bunch!

Marco

pavel-demin, 2023-08-10:
I have just noticed that your board is STEMlab 125-10.

The correct project for this board is `pulsed_nmr` and the commands to build the project are
```
set argv [list pulsed_nmr xc7z010clg400-1]
source scripts/project.tcl
```

closed 2023-08-10 by pavel-demin

marcomeixner, 2023-08-21:
Thank you so much for your quick reply!!

I now ended up here:

start_gui
cd C:/redPitaya/red-pitaya-notes-master/
set argv [list pulsed_nmr xc7z010clg400-1]
pulsed_nmr xc7z010clg400-1
source C:/redPitaya/red-pitaya-notes-master/scripts/project.tcl

Then it runs quite far and even the Diagram is created, until the following errors:
--------------------------------------------------------------------------------
ERROR: [BD 5-390] IP definition not found for VLNV: pavel-demin:user:axis_red_pitaya_adc:* 
ERROR: [Common 17-39] 'create_bd_cell' failed due to earlier errors.

    while executing
"create_bd_cell -type ip -vlnv $cell_vlnv $cell_name"
    (procedure "cell" line 2)
    invoked from within
"cell pavel-demin:user:axis_red_pitaya_adc adc_0 {
  ADC_DATA_WIDTH 14
} {
  aclk pll_0/clk_out1
  adc_dat_a adc_dat_a_i
  adc_dat_b adc_dat_b_i
  adc_..."
    (file "projects/pulsed_nmr/block_design.tcl" line 71)

    while executing
"source projects/$project_name/block_design.tcl"
    (file "C:/redPitaya/red-pitaya-notes-master/scripts/project.tcl" line 68)
update_compile_order -fileset sources_1
--------------------------------------------------------------------------------

I do have WSL running. You think I need to switch to that now?

Thank you so much!!

pavel-demin, 2023-08-21:
It looks like you are still missing the first step of running the `build-cores.tcl` script.

marcomeixner, 2023-08-22:
Ah ok thank you!

When I run this command:
source C:/redPitaya/red-pitaya-notes-master/helpers/build-cores.tcl

and emtpy the his folder before:
C:\redPitaya\red-pitaya-notes-master\tmp

Vivado  jumps back an forth until I get this error:
ERROR: [Common 17-53] User Exception: Project already exists on disk, please use '-force' option to overwrite: 
     C:/redPitaya/red-pitaya-notes-master/tmp/cores/axis_fifo.sim

I made a video in case you are interested in the details:
https://cloudstore.zih.tu-dresden.de/index.php/s/DyNmdEAZBgbJBrr

If this gets too annoying just let me know and I will switch to WSL.

In any case - thanks a lot for your help!

pavel-demin, 2023-08-22:
Thank you for the video.

I have just checked that there are indeed two `axis_fifo` entries in the version of the `build-cores.tcl` script that you are using. It seems that Vivado on Windows creates additional `.sim` directories for cores it does not create on Linux.

I have just added the `.sim` directory to the `delete` command in the `core.tcl` script. I have also changed the `build-cores.tcl` script last week. The current version does not contain the two `axis_fifo` entries.

Could you try updating your version of `red-pitaya-notes`? I think the latest version should work better for you.

pavel-demin, 2023-08-22:
I think that you are almost there.

The order of the commands is important. The `set argv ...` command must be run before running the `project.tcl` script:
```
set argv [list pulsed_nmr xc7z010clg400-1]
source C:/redPitaya/red-pitaya-notes-master/scripts/project.tcl
```

See [this comment](https://github.com/pavel-demin/red-pitaya-notes/issues/1106#issuecomment-1671901967) for more details.

The `build-cores.tcl` script only needs to be run once to build the cores in the `tmp` directory. After that, the `project.tcl` script can be run multiple times if multiple projects need to be generated.

marcomeixner, 2023-08-25:
Great, in that order everything now seems to work, thank you!

I now without error did this:
- openend **pulsed_nmr.xpr** with Vivado
- clicked: "Generate Bitstream"
- copied the resulting system_wrapper.bit to the Pitaya
- and on the Pitaya I ran it via: cat /root/pulsed-nmr.bit > /dev/xdevcfg

However, how do I know if it works?
Is there a documentation of how to connect the Pitaya to the other components?

Thank you!

pavel-demin, 2023-08-25:
> However, how do I know if it works - meaning if it acually runs on the Pitaya without error?

When I test it, I connect OUT1 to IN1 and try to get the signal envelope plot as shown in the screenshot in the notes. Maybe you could try something similar.

> And: Is there a documentation of how to connect the Pitaya to the other components?

For me, it is still an "in progress" project that I have never tested with real NMR hardware. So I do not have any documentation about the connection to other components and the software is still very basic.

I know that Dr. Beau Webber uses the part of the code that runs on the Red Pitaya with his hardware and software. The article at the following link contains information about this system:
https://www.mdpi.com/2673-8023/3/2/28

Figure 6 in this article shows the hardware and software modules, with the connections between them.

marcomeixner, 2023-10-03:
I finally found the time to try out the build you were so kind to help me make.

Here is a screenvideo of me trying to connect the client to the server: 'Server-to-Pitaya_03-10-2023_19-02-15.mp4'
https://datashare.tu-dresden.de/s/DyNmdEAZBgbJBrr

Somehow the connection is refused. I also tried starting ' pulsed_nmr.py' with sudo rights, but no luck.

Do you have any clue what might be the problem?

Thank you so much for any input on this!

pavel-demin, 2023-10-04:
In the video you configure the FPGA but you do not start the server ([pulsed-nmr.c](https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/pulsed_nmr/server/pulsed-nmr.c)). Since the server is not running, the client cannot connect to it.

marcomeixner, 2023-10-09:
Ah ok thank you!

It has been a while since I used c, but here is what happens when I try to compile the file on the Pitaya:
https://datashare.tu-dresden.de/s/DyNmdEAZBgbJBrr
start_nmr_server_on_pitaya_ScreenVideo_09-10-2023_21-54-40.mp4

pavel-demin, 2023-10-09:
Try to compile the server code using the `make` command. This is what I use.

Out of curiosity. What is the reason behind using this most complicated way of running this application? Would not it be quicker and easier to follow the getting started instructions and have the pulsed NMR application running in minutes?

Today is the two months anniversary of this issue.

marcomeixner, 2023-10-10:
Sorry to bother you so much with this.

I thought I am trying the most straightforward way to start the server. I get the same error with 'make' btw.

pavel-demin, 2023-10-10:
I think that the most straightforward way to start the server is described in the getting started instructions at [this link](http://pavel-demin.github.io/red-pitaya-notes/pulsed-nmr). Here is a copy of the part of instructions related to the server:

- Download [SD card image zip file](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20231004/red-pitaya-alpine-3.18-armv7-20231004.zip) (more details about the SD card image can be found at [this link](http://pavel-demin.github.io/red-pitaya-notes/alpine/)).
- Copy the contents of the SD card image zip file to a micro SD card.
- Optionally, to start the application automatically at boot time, copy its start.sh file from apps/pulsed_nmr to the topmost directory on the SD card.
- Install the micro SD card in the Red Pitaya board and connect the power.

marcomeixner, 2023-10-10:
Ok. Thanks a lot for all your help!