pavel-demin/red-pitaya-notes #1109: Continuous signal acquisition in parallel with processing

brhb, 2023-10-23:
Hello Pavel,

I hope this message finds you well. I wanted to express my appreciation for your contributions and projects.

### Description of the setup:

 * Device: stemlab 125.14
 * SD card image: alpine
 * Application: adc_record , adc_test_server
 * Other relevant information:

### Description of the problem:

I'm currently working on a real-time signal acquisition project that involves capturing signals with a frequency exceeding 2 MHz. The specific events I need to detect, which I'll refer to as 'peaks,' occur at a frequency of _alpha_ microseconds.

### Steps to reproduce the problem:

 1. Continuous Signal Acquisition: I need to continuously acquire signals in real-time.
 2. Peak Detection and Data Transfer: In parallel with signal acquisition, I aim to detect these 'peak' events and save the samples surrounding them. Subsequently, I plan to transfer this buffered data to a PC for further analysis.

Thank you in advance for your assistance.

Best regards,
brhb

pavel-demin, 2023-10-23:
I am not sure if I understand what you expect from me regarding this issue.

I think that peak detection and data transfer can be achieved by combining [axis_pulse_height_analyzer](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_pulse_height_analyzer.v) and [axis_oscilloscope](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_oscilloscope.v).

An example project that analyzes peaks using [axis_pulse_height_analyzer](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_pulse_height_analyzer.v) is [Multichannel Pulse Height Analyzer](http://pavel-demin.github.io/red-pitaya-notes/mcpha/).

If you want to transfer data from the Red Pitaya board to a computer, then you will need to start by calculating the required data throughput and making sure that it matches the capabilities of the Red Pitaya card.

brhb, 2023-10-23:
Thank you for the fast response.

I am actually new to TCL and FPGA programming, so I've divided the work into two steps:

Ensure continuous signal acquisition: At this stage, I'm using the adc_recorder and its block_design to acquire the signal continuously. I have removed the blocks "bcast_0," "cic_0," "cic_1," "comb_0," and "fir_0" to achieve the full sampling rate. However, there is still an issue with discontinuities due to buffer underflows. My idea is to modify the sampling rate to 10 MHz and stream the signal without interruption. So, my questions are: How can I adjust the sample rate, and how can I eliminate the sleep time?

Peak detection in C threading: Based on your response, it seems that this is possible with the FPGA axis_hub. Could you please guide me on how to integrate this module with adc_recorder?

Thank you in advance."

pavel-demin, 2023-10-23:
I am afraid that I cannot help with these kinds of questions.

closed 2023-10-23 by pavel-demin