pavel-demin/red-pitaya-notes #1110: Partial Reconfiguration on Red Pitaya

zzyzzy42, 2023-10-25:
### Description of the setup:

 * Device: 122-16
 * SD card image: Factory image or your custom image
 * Application: Partial reconfiguration for a research project
 * Other relevant information:

### Description of the problem:

Hello Pavel,

I hope you are well!

I doubt this is the right place to ask, but I'm wondering if you have experience in doing partial reconfiguration on Red Pitaya.

We tried to first upload the bitstreams (static and partial reconfigurable) to Red Pitaya using _scp_ command and naively used _cat filename > /dev/xdevcfg_ to configure the PR region without any success. There must be a right way to program the PR regions in the FPGA.

We have achieved partial reconfiguration on an evaluation board but have no clue how to do it on Red Pitaya.

We are wondering if you have any advice or suggestions.

Thank you so much for your help!

### Steps to reproduce the problem:

 1. Generate bitstreams (static and partial) for Red Pitaya on a Linux device.
 2. Copy the bitstreams to a Red Pitaya using scp command through ssh.
 3. cat filename > /dev/xdevcfg to program the FPGA.
 4. Command works with complete bitstreams but not PR bitstreams.

pavel-demin, 2023-10-25:
According to the Xilinx documentation at [this link](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841851/XAPP1231+-+Partial+Reconfiguration+of+a+Hardware+Accelerator+with+Vivado+Design+Suite), the is_partial_bitstream flag should be set to 1 before sending PR bitstreams.

So I think that the following commands should be used for partial reconfiguration:
```
echo 1 > /sys/devices/soc0/axi/f8007000.devcfg/is_partial_bitstream
cat pr.bit > /dev/xdevcfg
```

I have never done the partial reconfiguration myself so I am curious if these commands really work. Please let me know.

zzyzzy42, 2023-10-26:
Hi Pavel,

Thanks a lot for your help!

Unfortunately, it is not working. I can feel that you are very close because it is definitely doing something. Could you please take a look at the screenshot I attached?

[image]

I used the echo command from the link you shared because _axi_ does not exist under _soc0_, but amba does. Why did you make this change?

I can share my bitstreams with you if needed. Thank you soooo much again!

Best,
ZZ

pavel-demin, 2023-10-26:
Thank you for the test.

> I used the echo command from the link you shared because axi does not exist under soc0, but amba does. Why did you make this change?

The `amba` directory works with older versions of the Linux kernel. The `axi` directory works with more recent versions of the Linux kernel. I am currently using Linux kernel 6.1 and it only shows the `axi` directory.

I see that the [PDF version of XAPP1231](http://www.xilinx.com/support/documentation/application_notes/xapp1231-partial-reconfig-hw-accelerator-vivado.pdf) has a tiny bit more information about the `is_partial_bitstream` flag and the exact commands that you are using.

Have you tried configuring the FPGA via the JTAG interface to make sure that the problem is not with the bitstream files?

zzyzzy42, 2023-10-27:
Hi Pavel,

Thank you so much for the information!

I have just ordered a JTAG adaptor. We wanted to save some money and did not buy one, but I guess we have to have one haha.

I will let you know if I find out anything.

Thanks again! Have a nice rest of your day!

Best,
ZZ

zzyzzy42, 2023-11-14:
Hello Pavel,

I hope you had an awesome weekend! Sorry for the long wait! Idiot I used a wrong pinout table for my JTAG adaptor...

I configured the FPGA via JTAG interface and was able to perform partial reconfiguration on RP. They were the same bitstreams I used when I did
`cat pr.bit > /dev/xdevcfg`
`echo 1 > /sys/devices/soc0/amba/f8007000.devcfg/is_partial_bitstream`

Do you have any other idea on what might be causing the problem?

I really appreciate all your help! I understand this is might not be the right place to ask, but you were still willing to help!

Best,
ZZ

pavel-demin, 2023-11-14:
Thank you for the update.

I have checked the source code of the [xdevcfg](https://github.com/pavel-demin/red-pitaya-notes/blob/master/patches/xilinx_devcfg.c) driver and I do not see any particular problem that could be related to the partial reconfiguration.

I am out of ideas at the moment.

I have looked at several DFX (partial reconfiguration) tutorials and at the moment I think that I could manage to follow the one at [this link](https://github.com/Xilinx/Vivado-Design-Tutorials/tree/2023.1/Design_Flow_Tutorials/IP_Integrator/Lab_2). I think that if I can combine this tutorial with the playground project, then I will have a way to reproduce the problem.

zzyzzy42, 2023-11-15:
Hi Pavel,

Thanks a lot for your help! I will keep trying!

Have a great rest of your week!

Best,
ZZ

pavel-demin, 2023-11-15:
I have managed to follow the Lab 10 tutorial from UG947 and after some struggling with pblock constraints I have finally obtained partial reconfiguration bitstreams.

However, I am unable to reproduce your problem. Partial reconfiguration works with my SD card image:
```
# cat system_wrapper.bit > /dev/xdevcfg 
# ./test 6 7 
0x0000002a 42
# echo 1 > /sys/devices/soc0/axi/f8007000.devcfg/is_partial_bitstream
# cat system_i_rp1_rp1rm2_inst_0_partial.bit > /dev/xdevcfg 
# ./test 6 7 
0x0000000d 13
# cat system_i_rp1_rp1rm1_inst_0_partial.bit > /dev/xdevcfg 
# ./test 6 7 
0x0000002a 42
```

The `rp1rm1` bitstream multiplies the arguments of the test program and the `rp1rm2` bitstream adds them.

Maybe your problem could be solved by using the latest version of my SD card image.

pavel-demin, 2023-11-16:
Here is a link to the source code for my test project:

https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/dfx_test_122_88

Since the partial reconfiguration works with the latest version of my SD card image, there is nothing to fix. Closing this issue.