pavel-demin/red-pitaya-notes #1112: Question about application for GHz dual-channel downconversion receiver for GHz VNA

gsteele13, 2023-11-04:
### Description of the setup:

 * Device: None yet
 * SD card image: None yet
 * Application: GHz VNA receiver
 * Other relevant information:

### Description of the problem:

I'm looking into implementing a GHz VNA using a red pitaya as the IF digitizer / demodulator to replace the commercial GHz VNAs (eg. PNA5222A) we have in the lab. 

I've been looking around and think that the [SDR receiver](https://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/) could be a great building block for this, but then with two channels that share a common single digital IQ demodulation reference to achieve phase alignment. Here is a block diagram with my full idea:

[image]

I'd like to stream the IQ data from the RP straight into python where I can further decimate / filter (if needed) to increase SNR (we often use a 1 Hz IFBW in our data acquisition...). We need the two channels to calibrate out the non-flatness of the generator power vs freq, and also to measure the randomized phase alignment we get on the two MW generator channels whenever a frequency is changed, along with calibrating and correcting for imperfections in the downconversion mixers. 

*(for RF nerds: The down conversion is vulnerable to image leakage, but that is not a problem in principle since only one frequency is coming in on their RF ports...this is also how our expensive commercial VNAs are implemented...)*

I guess my question is if there is a possibility to implement this idea using the code in this repo, or maybe with some modifications?

My requirements are:

* Two SDR receivers running from a single digital ref stream so that the IQs on both inputs are phase aligned
* Ability to frequency lock the clock of the RP to the 10 MHz ref from my GHz generator
* Library for getting the IQ streams into my python software 

And some things that would be nice would be, on the longer term, could be:

* More filtering / decimation of the IQ streams on the RP to reduce CPU load / complexity of my python program
* IQ Trace acquisition on the RP started by a hardware trigger to align to a programmed frequency sweep on my generator for future applications of very fast traces 

I'd be very interested to hear what you think, and can also donate my own time (limited) and perhaps also time of a university programmer to help 

Thanks in advance!
Gary

pavel-demin, 2023-11-04:
I think the trickiest part would be locking the frequency to the 10 MHz reference.

Why (re)restart from the SDR receiver if there is already a working [VNA](http://pavel-demin.github.io/red-pitaya-notes/vna) in this repository? It already has two synchronous SDR receivers and a Python code example.

gsteele13, 2023-11-04:
> Why (re)start from the SDR receiver if there is already a working [VNA](http://pavel-demin.github.io/red-pitaya-notes/vna) in this repository? It already has two synchronous SDR receivers and a Python code example.

Excellent question! I did not realise the current 50 MHz VNA had two receiver! That is totally 100% the place to start then. Thanks! 

> I think the trickiest part would be locking the frequency to the 10 MHz reference.

I guess since I will not be using the RP outputs on my VNA, I could use one of the RP outputs to output a 10 MHz reference and use the generator in external REF mode, that would probably work, I'll dig into the code as soon as I have some free time

Cheers,
Gary

pavel-demin, 2023-11-05:
If you do not use the outputs, then I think you do not need additional frequency synchronization. I think that the phase difference between IN1 and IN2 signals does not depend on the parameters of the local oscillator used for mixing inside the SDR block on your diagram.

gsteele13, 2023-11-05:
Agree, as long as only a single LO is used in the SDR block for both inputs, then we should be good! I am looking forward to building it and trying it :)

mgarrett1955, 2024-10-22:
I am also looking for a phase coherent reciever for ch1 and 2 for an vector power meter, recieving voltage and current. It seems that the VNA example is suitabl for what I need. I was hoping to strem the signal into gnuradio (The singal in the sdr_tranciever, do not seem to be phase coherent?)  Did your evaluation of phase coherence work out??

gsteele13, 2024-10-22:
In the end, I got a team of electrical engineering students last semester
who build an custom iq demodulator and streaming server for our application

We are doing some last testing now but hope to make an alpha version of the
codebase and hardware specs public in coming month

closed 2024-12-02 by pavel-demin