pavel-demin/red-pitaya-notes #1116: Multichannel Pulse Height Analyzer's constraints related to the pulse duration?

danielbon, 2023-12-04:
Device: STEMlab 125-10
SD card image: last
Application: Multichannel Pulse Height Analyzer
Other relevant information:

We are working with pulses with 300 ns duration and the energy resolution is worse than working with longer pulses (e.g. 1000 ns). Does the Multichannel Pulse Height Analyzer have any constraints related to the pulse duration?

pavel-demin, 2023-12-04:
It would be helpful if you could post a screenshot showing the pulse shape on the oscilloscope tab of the control program. Without seeing the pulse shape, I do not understand what part of the pulse corresponds to 300 ns.

I use this application with shaping amplifiers with a shaping time of around 1 us and a total pulse duration of around 6-7 us.

If you look at the diagram in the MCPHA application notes, you can see that the maximum sample rate at the input to the pulse height analyzer module is 31.25 MSPS. I would say that a Gaussian-shaped pulse should be composed of at least a few dozen samples for its height to be measured with reasonable accuracy.

300 ns corresponds to approximately 9 samples at the sample rate of 31.25 MSPS. If the total pulse duration is 300 ns, it is indeed too short for this application.

1000 ns corresponds to approximately 31 samples at the sample rate of 31.25 MSPS. This number of samples seems to be close to a reasonable number of samples.

A possible solution to working with shorter pulses could be to remove the CIC filter. It can be done by modifying the source code and rebuilding the bistream file.

danielbon, 2023-12-05:
Thanks again for the explanations. The pulse has a double exponencial shape with a decay constant of about 40 ns. The detector is composed by a GaGG scintillator coupled to a Silicon photomultiplier. We will acquire a pulse to upload here, but the easiest option for us would be to use a shaping circuit since we are not familiar with the code. If the algorithm can get only 9 samples in this case, the precision of the peak value would be very limited, indeed.

pavel-demin, 2023-12-05:
> but the easiest option for us would be to use a shaping circuit since we are not familiar with the code.

Here is a link to a bitstream file without CIC filter:

https://www.dropbox.com/sh/5fy49wae6xwxa8a/AADO_iiCdWNY2mEW7afPHgaUa/mcpha/mcpha_no_cic.bit?dl=1

Just copy it to `apps/mcpha/mcpha.bit` and restart the MCPHA application on the Red Pitaya board.

danielbon, 2023-12-07:
Dear Pavel,
Sorry for the delay and thanks for the bitstream file. We are still testing it. I let you know once we got some good results.

closed 2023-12-25 by pavel-demin

danielbon, 2024-01-10:
Dear Pavel, 
In the end, we decided to use the original code and a amplifier to have a proper signal as input. Thanks for the support, anyway.