pavel-demin/red-pitaya-notes #1117: Please add VERSION to the wspr uploads <ENHANCEMENT>

waterwin, 2023-12-13:
Currently spots are not reported into the wsprnet.org database with a software version. Most (all) other types of decoding software do.

Suggested changes to decode-wspr.sh

VERSION=STEMlab-PD (for the 125 MHz / 14 bit)
VERSION=SDRlab-PD (for the 122.88 MHz / 16 bit)

curl -sS -m 30 -F allmept=@$ALLMEPT -F call=$CALL -F grid=$GRID -F version=$VERSION http://wsprnet.org/post > /dev/null

Thank you,
Erwin

pavel-demin, 2023-12-13:
I prefer that my software reports as little information about its users as possible.

Since this version field is optional and not used by wsprnet.org, I see no reason to add it to my version of the script and I do not think it should be used to identify my software.

Decoding is done by the WSPR decoder from WSJT-X. I recently updated it with changes from version 2.6.1. I think this is the version that should be reported.

pavel-demin, 2023-12-13:
I have just added the VERSION variable with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/a31d685073849c73f9ea404227dfdcfc9948f881). It is empty by default and it is up to the operator to set it to the value they want.

pavel-demin, 2023-12-13:
I have just checked the versions reported in the last 24 hours and I prefer the format used by DC1RDB (`RP125-14` or `RP122-16`).

waterwin, 2023-12-13:
That is perfect.

pavel-demin, 2023-12-13:
I have updated the version information with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/12241912f9c22503f5faa87057b4cbb25b1aacf9).

closed 2023-12-13 by waterwin