pavel-demin/red-pitaya-notes #1119: Question about fir for 122.88 hpsdr receiver with 7010 FPGA (TRX-DUO)

enthru, 2023-12-22:
Hello!

It's impossible to generate hpsdr receiver bitstream for this FPGA just using approach like for building bitstream for hpsdr transceiver (_make NAME=sdr_transceiver_hpsdr_122_88 bit_) because there is no enough space in dsp48e block with fir filter for 7020 FPGA.

Just want to ask you about advice or possible solution how to get it work without calculation of new fir and changing a lot in rx module keeping the same or not much less count of the receivers.

pavel-demin, 2023-12-22:
The projects with the `122_88` suffix are for the Red Pitaya STEMlab 122.88-16 SDR and SDRlab 122-16 boards with a larger FPGA.

The corresponding project for boards with Zynq 7010 is `sdr_transceiver_hpsdr`. The command to build bitstream for this project is
```
make NAME=sdr_transceiver_hpsdr bit
```

closed 2023-12-22 by pavel-demin

enthru, 2023-12-22:
Yep, I know I'm trying to build 122.88 version for  7010 fpga like this:

https://github.com/pavel-demin/red-pitaya-notes/issues/1091

pavel-demin, 2023-12-22:
I see.

The command in #1091 no longer works with the current version of the code. The `sdr_transceiver_hpsdr_122_88` project now has additional receivers from the `sdr_receiver_hpsdr_122_88` project and the 7010 chip does not have enough resources for these additional receivers.

To make the `sdr_receiver_hpsdr_122_88` project compatible with the 7010 chip, you will need to reduce the sample rates (`MINIMUM_RATE` and `SAMPLE_FREQUENCY`) of the CIC and FIR filters by a factor of 2. The coefficients can be copied from the `sdr_receiver_122_88` project:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/projects/sdr_receiver_122_88/rx.tcl

The C code also needs to be changed to remove the 384k sample rate.

enthru, 2023-12-23:
Thanks. Also was forced to remove RX1 in block design, it builds, but for some reason I'm unable to get it work - see just something like dc on the center frequency. Trying to figure out why :)

enthru, 2023-12-23:
Maybe did some mistake in server file, just made changes in C code from scratch and recompile it again and it works.