pavel-demin/red-pitaya-notes #1120: The MCPHA.exe could not establish connection to the IP address

cztee, 2023-12-25:
### Description of the setup:

 * Device: STEMLAB 125-14 4-input
 * SD card image: Latest Beta 2.0 (2.00-23)
 * Application: MCPHA
 * *Relevant information: The red pitaya is connected directly to router.

### Description of the problem: I followed the instruction provided in one of your replies on the forum: wget the mcpha.zip from Onedrive and unzip in "/opt/redpitaya/www/apps". Everything works, the icon showed up, I could establish a connection to the red pitaya via web and ssh (ssh.png and web_mcpha.png). I also downloaded the "mcpha-win32-20190527.zip" as instructed on the webpage. However, when I was trying to connect to the red pitaya via the same IP address in the mchpa.exe, I have the cannot connect to IP issue (error_cannot_connect.png). I tried with the scope application present in the red pitaya and it seems like there's no problem with the i/o port (regardless of which input port). Can you please advice what are the other things I can check? I think the MCPHA is amazing and my lab would like to use it instead of using LabVIEW. Thank you.

[image]
[image]
[image]
[image]

pavel-demin, 2023-12-25:
I recommend using the SD card image from this repository with the new control program.

Recently, Red Pitaya developers announced that applications from the application marketplace do not work with the new versions of their SD card image:
​
https://forum.redpitaya.com/viewtopic.php?f=8&t=25198&start=10

Maybe your problem could be solved by using an older version of the SD card image provided by Red Pitaya developers.

pavel-demin, 2023-12-25:
I have just realized that you are using a 4-input version of the Red Pitaya board. The connections between the ADC and the FPGA on this version of the board are quite different from those on the 2-input version of the Red Pitaya board.

Currently, I do not have a version of the MCPHA application adapted to the 4-input version of the Red Pitaya board.

closed 2023-12-25 by pavel-demin