pavel-demin/red-pitaya-notes #1121: EDIMAX EW-7811UN V.1 still supported?

dl5dla, 2023-12-26:
Red Pitaya STEMlab 125-14
red-pitaya-alpine-3.18-armv7-20231004.zip
EDIMAX EW-7811UN V.1

Hi,
in the past I already had installed the red-pitaya-alpine software and setup the FT8 transceiver application. This was working with LAN and also with wifi connection using the EDIMAX EW-7811UN V.1.
Now I have setup the Red Pitaya again with the newest release (3.18) and the same EDIMAX wifi stick.

Problem:
The EDIMAX stick is not recognized, when I plug it in (no entry in dmesg or /var/log/messages or lsusb)
As result there is no wlan0 interface shown (ifconfig)
To ensure that the USB port is not defect, I plugged in a USB memory stick, which was shown in dmesg.

As the EDIMAX stick is a V.1 type with probably different chipset, I am wondering whether support for V.1 was removed meanwhile.

Thanks.
 Peter

pavel-demin, 2023-12-28:
The driver that I used for the RTL8192CU chip (https://github.com/pvaret/rtl8192cu-fixes) does not work with the latest Linux kernel versions and I had to remove this driver.

Currently enabled drivers are for the following USB/Wi-Fi chips:
- RTL8188EU
- RT5370
- AR9271

It seems that newer versions of [EDIMAX EW-7811Un](https://www.reichelt.com//wlan-adapter-usb-150-mbit-s-edi-ew-7811un-v2-p293505.html), [D-LINK DWA-121](https://www.reichelt.com/wlan-adapter-usb-150-mbit-s-d-link-dwa-121-p280977.html), [TP-Link TL-WN725N](https://www.reichelt.com/wireless-lan-nano-usb-adapter-150-mbit-s-tplink-tl-wn725n-p123963.html), [TP-Link TL-WN722N](https://www.reichelt.com/wireless-lan-usb-adapter-150-mbit-s-detachable-aerial-tplink-tl-wn722n-p125227.html) and other similar 150 Mbps USB/Wi-Fi adapters use the RTL8188EU chip. So maybe it is time to update your USB/Wi-Fi adapter.

dl5dla, 2023-12-29:
Thanks for clarification. I will go with the TP-Link TL-WN725N then.

closed 2024-02-17 by pavel-demin