pavel-demin/red-pitaya-notes #1128: Minimal project with fpga-server -client structure with  hub interface, a ADC interface, and a DAC interface

marcomeixner, 2024-02-29:
Hey Pavel,

I am trying to understand the pulsed_nmr project.

I figured the template project in the last link you showed me would be a good start, but it does not have the fpga-server-client structure. I tried to use the server and the client from the pulsed_nmr project, and it runs, but the fifo appears to be empty.

Therefore I was wondering which project you would recomment as the most minimal one to start with, that does have a hub interface, a ADC interface, and a DAC interface - but also has a server and a client file.

Thank you so much!
Marco

pavel-demin, 2024-02-29:
I think that the [playground](http://pavel-demin.github.io/red-pitaya-notes/playground/) project is the minimal project with all the elements that you are looking for.

marcomeixner, 2024-02-29:
Great, thank you!!

closed 2024-03-01 by pavel-demin