pavel-demin/red-pitaya-notes #1130: Issue with Redpitaya working using GNU Radio

xL1n, 2024-03-28:
### Description of the setup:

 * Device: Redpitaya STEMlab 125-14
 * Application: using redpitaya as SDR transceiver working with GNU Radio 

### Description of the problem:
I am new to GNU Radio and Redpitaya, My intention is to transmit some real binary signal.
[image]
I start by trying the simple feedback loop which connected OUTPUT 1 port to IN 1 port using a SMA Cable, but each time when I transmit (run the flowgraph on GRC), I get different received signal with different "phases" (the reason why I am not sure if it's phase because I only transmit signal with real values, without any imaginary parts), and the received signal conatins some imaginary parts with real parts (complex signal, shown in red and blue at Rx), and I tried many times, just run the flowgraph without changing anything, each time the magnitude for Re and Im part are different and not following any sequences or patterns. 
I am confused with this, could someone help me with this, thanks. 
(the following screenshots are my 3 tests, for sample flowgraph, just run it 3 times, and gave me 3 different Rx signals)
[image]
[image]
[image]

pavel-demin, 2024-03-28:
I think that if you just need the signal amplitude, then it should be enough to add the "Complex to Mag" block right after the "Red Pitaya Source" block.

xL1n, 2024-03-28:
Thank you for reply, indeed, I tried with this to only get magnitude, however, I am not sure why each time the magnitude of complex and real part of received signal is different, I am just sending real time signal, I dont even understand where is that "complex signal" comes from.

pavel-demin, 2024-03-28:
The receivers and transmitters in the SDR transceiver application are completely independent and are not synchronized. So it is normal that there is no synchronization.

xL1n, 2024-03-28:
> The receivers and transmitters in the SDR transceiver application are completely independent and are not synchronized. So it is normal that there is no synchronization.

thanks for the information, I am still bit confused why each time receive different "signal" (each time the magnitude of complex and real time signal are different), and why there is even complex time signal if I am only sending real time signal. Could you please illustrate more, thanks

pavel-demin, 2024-03-28:
Please see the diagram showing digital down converters (DDC) and digital up converters (DUC) in the notes at the following link:

http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/

This diagram shows how complex signals are produced.

closed 2024-03-28 by pavel-demin

xL1n, 2024-03-28:
> Please see the diagram showing digital down converters (DDC) and digital up converters (DUC) in the notes at the following link:
> 
> http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/
> 
> This diagram shows how complex signals are produced.

Thanks, initially I thought it might need complete packet transmission for synchronization, Therefore, I tried the example for packet transmission on GRC Wiki 
[https://wiki.gnuradio.org/index.php/Packet_Communications?ver1=3.8#Version_3.8](url)
[image]
[image]

and again, used simple feedback loop (connecting OUTPUT1 to IN1 using SMA), however, I get strange reception that if looking at the constellation sink, it's "spinning" instead of showing BPSK. 

https://github.com/pavel-demin/red-pitaya-notes/assets/91786547/be46c3c2-dde6-4efd-9a0a-49b1b06219e6

Could you please help me with it, as I am bit confused, Thanks