pavel-demin/red-pitaya-notes #1131: Issue with Packet Transmission using Redpitaya and GNU Radio

xL1n, 2024-03-29:
### Description of the setup:

 * Device: Redpitaya STEMlab15-14
 * Application: Packet Transmissioin using redpitaya and GNU Radio 3.8

### Description of the problem:
Hi Pavel,
I appreciate you provide information for my last issue, thanks. 
I am still struggled with the redpitaya transmission. To be more specific, I used example of packet transmission in GRC Wiki 
[https://wiki.gnuradio.org/index.php/Packet_Communications?ver1=3.8#Version_3.8](url)
for hardware part, I just simply connect OUT1 to IN1 using SMA cable creating a simple feedback loop.
the flowgraph looks like following for transmit and receive in GRC:
[image]
[image]

However, at reception end, I get following constellation diagram, which is "rotating", (and if looking at received signal, each time the magnitude for real and imaginary parts are different, without following any pattern)

https://github.com/pavel-demin/red-pitaya-notes/assets/91786547/b0bb4feb-9046-4ff2-8b3f-f36781adffb4

This makes me really confused, and have no clue of what to do, could you please help me with this issue, please

pavel-demin, 2024-03-30:
I am not really a GNU Radio expert. I think that it is best to ask this kind of question on a forum for GNU Radio users.

I see that some blocks in your diagram use different sample rates and are directly connected without resampling. I think that setting all sample rates to correct values would be the first step in debugging your application.

xL1n, 2024-03-30:
> I am not really a GNU Radio expert. I think that it is best to ask this kind of question on a forum for GNU Radio users.
>
> I see that some blocks in your diagram use different sample rates and are [...]

Thanks for reply, I just found the where this issue comes from, I am transmit/receive baseband signal, but I set up the centre frequency of redpitaya sink and source to be 100k Hz, when I changed it to 0 Hz, it works.

closed 2024-03-30 by xL1n