pavel-demin/red-pitaya-notes #1132: Does old Vivado editon supported?

zkf0100007, 2024-04-11:
Hi,

I am using vivado 2020.1 now. As suggested in (https://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/), vivado 2023.1 is recommended. The install package of vivado 2023.1 is so big, I don't want to install it.
Is there a branch supporting vivado 2020.1?

Thanks!

pavel-demin, 2024-04-11:
I think it should be possible to use the 2020.1 version of the Xilinx tools with the latest code from this repository. Do you see any problems?

zkf0100007, 2024-04-12:
I followed the https://pavel-demin.github.io/red-pitaya-notes/led-blinker/.
When I run "make NAME=led_blinker all" , I got the error info as below:
`mkdir -p tmp/device-tree-xlnx-xilinx_v2020.2
tar -zxf tmp/device-tree-xlnx-xilinx_v2020.2.tar.gz --strip-components=1 --directory=tmp/device-tree-xlnx-xilinx_v2020.2

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
Makefile:81: recipe for target 'tmp/device-tree-xlnx-xilinx_v2020.2' failed
make: *** [tmp/device-tree-xlnx-xilinx_v2020.2] Error 2`

zkf0100007, 2024-04-12:
It seems that the makefile line 35 "DTREE_URL = https://github.com/Xilinx/device-tree-xlnx/archive/$(DTREE_TAG).tar.gz" is not correct. It should be DTREE_URL = https://github.com/Xilinx/device-tree-xlnx/tree/xlnx_rel_v2022.2

pavel-demin, 2024-04-12:
Thank you for the example. Looks like some of my scripts actually require the 2023.1 version of the Xilinx tools.

closed 2024-04-12 by pavel-demin

pavel-demin, 2024-12-11:
Thank for testing the `project.tcl` script with Vivado 2020.1. I have just fixed the problem that you found with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/d2ebdab1cb2beafe3877bf95c6747e134f913ac7).

I have also tried to build bitstream files for a couple of projects with Vivado 2018.2 and 2024.2 using the following commands:
```
make NAME=led_blinker bit
make NAME=sdr_receiver bit
```

These commands worked without any problems. So I think that it should be possible to build bitstream files for most of the projects in this repository with old versions of Vivado.