pavel-demin/red-pitaya-notes #1134: Bitstream Compilation Error for 122_88 projects.

wptaylor, 2024-05-10:
### Description of the setup:

 * Device: N/A
 * SD card image: N/A
 * Application: led_blinker_122_88, sdr_reciever_122_88
 * Other relevant information: Development setup followed as in [Development machine](https://pavel-demin.github.io/red-pitaya-notes/development-machine/) notes. Major variable difference is the Hypervisor method: I used Hyper-V instead of VirtualBox.

### Description of the problem:

Compilation of the bitstream for led_blinker_122_88 and sdr_reciever_122_88 fails at `synth_1` immediately following `impl_1`. 

Seems not to be a dev environment issue as led_blinker compiles consistently without issue. 

impl_1 runs in both cases for significantly longer than other steps (2+ mins for led_blinker, 20+ minutes for sdr_receiver).

synth_1 `runme.log` is clean, with no `ERROR`s. `impl_1.log` does not exist.

Have not attempted with most recent release. Will see if that fixes functionality.

### Steps to reproduce the problem:

 1. Clone active GitHub repo.
 2. Run `make clean <led_blinker_122_88 or sdr_receiver_122_88> all`

Same behaviour with `make NAME=sdr_receiver_122_88 PART=xc7z020clg400-1 bit` as specified verbatim [here](https://pavel-demin.github.io/red-pitaya-notes/sdr-receiver-122-88/). Same behaviour when run as root.

wptaylor, 2024-05-11:
Tried with the most recent release (20240204) for led_blinker_122_88 and got the same behaviour. Terminal output of the erroneous stage is pasted below. Here is the [runme.log](https://github.com/pavel-demin/red-pitaya-notes/files/15281310/runme.log) file from `tmp/led_blinker_122_88.runs/synth_1/`.

```
[Sat May 11 00:36:51 2024] Launched synth_1...
Run output will be captured here: /home/red-pitaya/red-pitaya-notes/tmp/led_blinker_122_88.runs/synth_1/runme.log
[Sat May 11 00:36:51 2024] Launched impl_1...
Run output will be captured here: /home/red-pitaya/red-pitaya-notes/tmp/led_blinker_122_88.runs/impl_1/runme.log
launch_runs: Time (s): cpu = 00:00:07 ; elapsed = 00:00:08 . Memory (MB): peak = 1336.332 ; gain = 21.879 ; free physical = 112 ; free virtual = 576
[Sat May 11 00:36:51 2024] Waiting for impl_1 to finish...
[Sat May 11 00:41:17 2024] impl_1 finished
ERROR: [Vivado 12-13638] Failed runs(s) : 'synth_1'
 'synth_1' run failed with below errors.
wait_on_runs: Time (s): cpu = 00:00:20 ; elapsed = 00:04:26 . Memory (MB): peak = 1342.270 ; gain = 5.938 ; free physical = 909 ; free virtual = 1176
ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.

    while executing
"wait_on_run impl_1"
    invoked from within
"if {[get_property PROGRESS [get_runs impl_1]] != "100%"} {
  launch_runs impl_1 -to_step route_design
  wait_on_run impl_1
}"
    (file "scripts/bitstream.tcl" line 6)
INFO: [Common 17-206] Exiting Vivado at Sat May 11 00:41:17 2024...
make: *** [Makefile:127: tmp/led_blinker_122_88.bit] Error 1
```

pavel-demin, 2024-05-12:
If it crashes without errors, it may be that there is not enough memory in your virtual machine or the problem is indeed with Hyper-V.

If you are using Windows, then it is also possible to use Windows Subsystem for Linux (WSL) and install Ubuntu Linux via the application store.

wptaylor, 2024-05-13:
Update: sdr_receiver compiles as intended. Is there a large memory requirement difference between 122_88 projects and others? If not, then it would seem to imply a specific issue with those compile chains (either in general or with my setup).

In any case, will try with a more powerful workstation/VM shortly. Will also try using WSL instead.

pavel-demin, 2024-05-14:
I have just checked that the memory usage is quite similar (between 2.5 and 3 GB) when building `led_blinker` and `led_blinker_122_88`. If I remember correctly, the virtual machine should have at least 4 GB of memory to build all projects.

wptaylor, 2024-05-14:
Interesting. I had allocated 8GiB+ (dynamic allocation) so I would be surprised if that was the issue. Will be checking with a Linux machine soon.

wptaylor, 2024-05-21:
Issue is fixed when using a full Ubuntu Linux boot with higher specs. Still unsure cause of error. Will try and determine if I get time later.

closed 2024-05-21 by wptaylor