pavel-demin/red-pitaya-notes #1136: RedPitaya / TRX-DUO ip config

VitGGGGGGGGGGGGGGGGGGGG, 2024-05-29:
### Description of the setup:

 * Device: TRX-DUO
 * SD card image: any for TRX-DUO compatible
 * Application:
 * Other relevant information:

### Description of the problem:
 Is it possible to make the network adapter configuration is located in the /apps directory of the SD card because, for example, I have a non-standard subnet adress and it takes time to change the configuration every time. It would be nice if it worked like in Pluto SDR?

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2024-05-30:
The wired interface is configured by default to request an IP address via DHCP. So if you connect your computer and the TRX-DUO board to a router that has DHCP functionality, then everything should work without any changes to the configuration files. Any wired or wireless router with Gigabit Ethernet ports and should do the job.

pavel-demin, 2024-05-30:
Thank you for the details. Personally, I try to keep things simple and avoid this kind of complicated network setup.

You also mentioned that you have to change the configuration every time. It should be enough to edit the configuration file once, then save the changes to the SD card using `lbu commit -d` and that is it.

If you want `dhcpcd.conf` to be located directly on the SD card, then I think this could be done with the following commands:
```
mv /etc/dhcpcd.conf /media/mmcblk0p1/dhcpcd.conf
ln -s /media/mmcblk0p1/dhcpcd.conf /etc/dhcpcd.conf
lbu commit -d
```

VitGGGGGGGGGGGGGGGGGGGG, 2024-05-30:
Thanks for the help! I'll try to make the configuration this weekend. if Thetis understood ip v6 addresses or accessing devices by their network names, I wouldn’t have such a problem because the device responds for  ip v6 or "rp-xxxxx" ping without "dances with a tambourine", but this is not useful to me.

VitGGGGGGGGGGGGGGGGGGGG, 2024-05-31:
Unfortunately, I'm doing something wrong again. My changes don’t  saved because for some reason the SD drive is readonly, although I just saved the changes to the network configuration in the editor
[image]

closed 2024-08-02 by pavel-demin