pavel-demin/red-pitaya-notes #1137: Build break due to apk-tools missing

howard0su, 2024-05-30:
### Description of the problem:
Fail to build due to cannot download apk-tools-static package from http://dl-cdn.alpinelinux.org/alpine/v3.18/main/armv7/ 

The file is updated to new version. I changed the version to match the download site but the image failed to boot.

pavel-demin, 2024-05-30:
I also noticed this problem. I think that this is caused by the following commit in apk-tools:

https://gitlab.alpinelinux.org/alpine/apk-tools/-/commit/042c5eb7680ffb2766c16f444c4ab75f7b86d5ca

This changed the way the hash of APKINDEX files is calculated. Currently, the latest 3.18 release contains an older version of apk-tools that is apparently incompatible with APKINDEX files generated by newer versions of apk-tools.

I think that this problem will be resolved once a newer 3.18 release appears.

At the moment I think that the easiest solution would be to switch to Alpine Linux 3.20 released a few days ago. I have just updated the scripts accordingly with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/67242df98a51add20f7fe59264b9220db60df963).

howard0su, 2024-06-01:
Thank you for quick turnaround. the fix works great!

closed 2024-06-01 by howard0su