pavel-demin/red-pitaya-notes #1138: Multichannel analyser

LodovicoLappetito, 2024-07-06:
### Description of the setup:

 * Device: Red Pitaya STEMlab 125-14
 * SD card image:
 * Application: Multichannel analyser
 * Other relevant information:

### Description of the problem:
Hello Pavel,
first of all thank you for your effort.
I am new at red pitaya, I tried to install your applications (I am interested in the MCA) but without success.
I am writing to you just to know if the MCA application is still compatible with the last version of the card (125-14)
Thank you
Lodovico

pavel-demin, 2024-07-06:
It would be helpful if you provided information about the SD card image that you are using.

If you follow the [getting started instructions](https://pavel-demin.github.io/red-pitaya-notes/mcpha/#getting-started-with-ms-windows) and use the [red-pitaya-alpine-3.18-armv7-20240204.zip](https://github.com/pavel-demin/red-pitaya-notes/releases/download/20240204/red-pitaya-alpine-3.18-armv7-20240204.zip) SD card image, then the MCPHA application should work with any version of the Red Pitaya STEMlab 125-14 board

LodovicoLappetito, 2024-07-06:
Hi Pavel, thank you for your response!
I am using a SD card preloaded with the last image from the producer.
With these commands (from redpitaya forum)
rw
cd /opt/redpitaya/www/apps
wget 'https://www.dropbox.com/sh/5fy49wae6xwxa8a/AACX0vv9VXs1U5-jT-oM6KOfa/bazaar/mcpha-1.0-20190527.zip?dl=1' -O mcpha-1.0-20190527.zip
unzip mcpha-1.0-20190527.zip
ro
I managed to install the application MCPHA and I see the icon in the webpage and the application start successfully.
I downloaded on my PC the control application but when I start MCPHA.exe and I try to connect to the redpitaya (with the correct IP address) I receive the error connection refused.
Lodovico

LodovicoLappetito, 2024-07-06:
I tried also with the linked zip file, I unzipped and I copied the files in a new FAT32 micro SD card but with that SD card the redpitaya does not work, maybe I do something wrong with the SD card.
thank you 
Lodovico

pavel-demin, 2024-07-06:
I no longer test my applications with the SD card images provided by Red Pitaya. Since they no longer want to support 3rd party applications, I am not interested in supporting their SD card images.

What you are doing with the SD card image from this repository seems OK. I do not see why you think that this does not work. You can find troubleshooting instructions at the following link:
http://pavel-demin.github.io/red-pitaya-notes/alpine/#troubleshooting

LodovicoLappetito, 2024-07-07:
Hello,
I managed to work with your Alpine SD card image, previously there was a network problem.
The MCPHA now is working! Thanks.
I have a question, I saved the histogram file, can you explain briefly the protocol of the file?
Thank you!
Lodovico

pavel-demin, 2024-07-08:
If I remember correctly, the histogram file is just a text file with each line containing the number of entries in the corresponding histogram bin.

Do you have any particular problems with the histogram file format?

LodovicoLappetito, 2024-07-08:
Hello, yes I managed to work on that txt file with the application Interspec, thanks!
Lodovico

closed 2024-07-08 by LodovicoLappetito