pavel-demin/red-pitaya-notes #1144: Question on Building a bootable SD card

zkf0100007, 2024-09-02:
I am using Vitis 2013.1 to run LED blinker example following the guide https://pavel-demin.github.io/red-pitaya-notes/led-blinker/
When I ran 'sudo sh scripts/debian.sh /dev/mmcblk0'
I found my TF card is recognized as /dev/sdX rather than /dev/mmcblkX, 
Could you tell me how to solve this problem?
Thanks!

pavel-demin, 2024-09-02:
Thank you for testing the getting started instructions.

If your SD card has a different path in your system, then I think that it should be enough to specify that path when running the `debian.sh` script.

However, I no longer use this command and create an SD card image before writing it to an SD card. So I would suggest to skip this command and go straight to the SD card image section.

I have just updated the getting started instructions and removed the `sudo sh scripts/debian.sh /dev/mmcblk0` command. I have also added some more useful example commands.

zkf0100007, 2024-09-02:
Thanks for your quick reply!
I will have a try following your suggestion.

closed 2024-09-07 by pavel-demin