pavel-demin/red-pitaya-notes #1145: Template Project, not programming

kevinamonteverde, 2024-09-04:
### Description of the setup:

 * Device: Red Pitaya STEMLab 125-14
 * SD card image: 2.04 - 35
 * Application: Template Project
 * Other relevant information:

### Description of the problem:

Hi, I'm building the template project from source as instructed by you. When the project is built, I open it in vivado and generate the bitstream, and scp the template.bit file to my board via a direct ethernet connection. Upon the command `cat template.bit > /dev/xdevcfg`, nothing happens. I assume the template project simply passes voltages at inputs to corresponding outputs, but nothing happens. Is the FPGA not programming or am I misusing the code? 

Thanks so much,
Kevin

### Steps to reproduce the problem:

 1.
 2.
 3.

pavel-demin, 2024-09-04:
Thank you for testing the `template` project and for the additional details about your setup.

It looks like you are not using the SD card image from this repository but an SD card image provided by Red Pitaya.

As far as I know, the Red Pitaya  developers have stopped providing the `xdevcfg` driver with their SD card images. I think that you will need to configure the FPGA using the method supported by Red Pitaya:
https://redpitaya-knowledge-base.readthedocs.io/en/latest/learn_fpga/3_vivado_env/tutorfpga2.html#reprogramming-the-fpga

kevinamonteverde, 2024-09-05:
Got it, thank you. Upon doing that and loading the file using the supported commands, I get the following error:

`root@rp-f0cd1e:~# fpgautil -b template.bit.bin
sh: 1: echo: echo: I/O error
BIN FILE loading through FPGA manager failed
`

pavel-demin, 2024-09-05:
I have never used the recent SD card images provided by Red Pitaya and their `fpgautil`. So I do not know why the command in their documentation does not work.

Actually, I do not understand why you need an extra step to generate bitstream. Normally it is generated by `make NAME=template bit`.

I can only recommend to try using the latest version of the SD card image from this repository. This is what I use to test my projects.

closed 2024-09-07 by pavel-demin

kevinamonteverde, 2024-09-11:
Apologies. Where in the repo is the SD card image you're referencing? Upon downloading the prebuilt image from http://pavel-demin.github.io/red-pitaya-notes/led-blinker/, the only img i could find, the red pitaya does not respond (only the power supply is on).

pavel-demin, 2024-09-11:
I am surprised that you managed to find the well hidden SD card image in the LED blinker notes and could not find the other one that is linked from all other notes and from the [releases](https://github.com/pavel-demin/red-pitaya-notes/releases). You can find the information about the other SD card image at [this link](http://pavel-demin.github.io/red-pitaya-notes/alpine).

The SD card image that you have found is also OK. It is normal that there are no blinking LEDs after booting the Red Pitaya board with this SD card image. Have you tried connecting to the Red Pitaya board via SSH while running this SD card image or checking the boot process using USB/serial console?

kevinamonteverde, 2024-09-16:
That's a very useful link. Thanks! Using the image you linked from http://pavel-demin.github.io/red-pitaya-notes/alpine, following the instructions (copying the files to a clean sd card, formatted as specified), I cannot connect to the red pitaya using `ssh root@rp-xxxxxx.local` using my board's mac address. I get the error message 'could not resolve hostname.' I reviewed the booting using the USB connection and it does not seem to boot. It does boot with the image given by RedPitaya in their website, so we aren't using the serial terminal improperly. My guess is that the way the files should be placed in the SD card using your image should be different, or there is in fact an image to be written to the sd card. We're not sure where to go from here.

pavel-demin, 2024-09-16:
I have intentionally made the installation instructions as simple as I could: simply take a freshly unpacked factory formatted micro SD and unpack the .zip file to it. There is no special way to do it.

kevinamonteverde, 2024-09-16:
That's what's so odd. Every step has been followed, yet the booting process is not occurring. Yet it occurs with the FPGA image from Red Pitaya, which means the problem either arises from these files or the way they are being placed in the SD card. If the problem is not how they are placed in the card, it must be the files. Like you said, there is no special way to do it. We are simply unpacking the .zip to a formatted sd card.

pavel-demin, 2024-09-17:
Whenever a problem like this occurs, there is a problem with the SD card, either it is not formatted properly or it is defective.

If you want me to try to figure out what the problem is, then please provide me with more information. Lists of commands and screenshots can be very helpful.

With the level of information that you are currently providing, I can only guess how you are formatting the SD card, how you are copying the files, etc.

kevinamonteverde, 2024-09-18:
> Whenever a problem like this occurs, there is a problem with the SD card, either it is not formatted properly or it is defective.
>
> If you want me to try to figure out what the problem is, then [...]

Hi, it seems that I get the same error as before (the I/O error upon bitstream loading) when trying to do the most basic fpga tutorials on the red pitaya website, following all the directions exactly. It seems that the problem might be with my board, as I can't find a similar issue online.

Thanks for all your help. Seems there was never a problem with this repository, so sorry for opening the issue. I'm going to consult the rp forums for advice.

pavel-demin, 2024-09-18:
It does not look like a problem with the board. Have you tried changing the micro SD card?