pavel-demin/red-pitaya-notes #1146: errors on building from source using vitis running on ubuntu

zkf0100007, 2024-09-06:
I installed ubuntu 18.04 virtual machine on vmware, and installed vitis 2023.1 on ubuntu.
When I building from source following https://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/ 
I could generate bitstream using 
`make NAME=sdr_receiver bit`
but some errors occured:
`source helpers/build-all.sh`
the error shows as following:
`qemu: Unsupported syscall: 397`
I wonder if I should use Debian insted of Ubuntu?
Could you give more detailed description on how to buildinf from souce?
Thanks!

pavel-demin, 2024-09-06:
I do not see any problem with the commands that you are using. I am using the exact same commands. Only I am using Debian 12 as the operating system on my computer.

Ubuntu should be OK, but 18.04 is a very old version. I would try a newer version. For example 22.04.

zkf0100007, 2024-09-07:
I changed OS from Ubuntu 18.04 to Debian 12 as suggested.
Bootimage generated successfully!

Thank you very much!

closed 2024-09-07 by pavel-demin