pavel-demin/red-pitaya-notes #1151: could not found 'red-pitaya.apkovl.tar.gz'

zkf0100007, 2024-10-02:
I am using VIVADO 2023.1 runnig on Debian 12.
I followed all your steps , using the following command:
`source helpers/build-all.sh`

Finally, I got red-pitaya-alpine-3.20-armv7-20240918.zip
When I unzip it, there are only 
apps
cache
wifi
boot.bin
modloop

there's no 'red-pitaya.apkovl.tar.gz' in the red-pitaya-alpine-3.20-armv7-20240918.zip

Could you tell me why?

Thanks!

pavel-demin, 2024-10-02:
A possible source of this problem could be that the versions of some packages used by [alpine.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/alpine.sh) are no longer available in the Alpine Linux repository and the script needs to be updated to use newer versions of these packages.

I have just updated [alpine.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/alpine.sh) and [alpine-project.sh](https://github.com/pavel-demin/red-pitaya-notes/blob/master/scripts/alpine-project.sh) with this [commit](https://github.com/pavel-demin/red-pitaya-notes/commit/6b08e8c009b0286bef2c61d2eb2641f98abf69a0).

zkf0100007, 2024-10-03:
OK，Thank you very much.
I solve the proble.
It is due to my poor internet connection, failed to download some file.
You konw, we have GFW in our country, haha.

closed 2024-10-03 by pavel-demin