pavel-demin/red-pitaya-notes #1152: Can't start adc_recorder again, ssh broken

mrengineer, 2024-10-06:
### Description of the setup:

 * Device: TRX-DUO
 * SD card image: latest from your repo
 * Application: adc_recorder
 * Other relevant information:

### Description of the problem:
After finishing programm adc_recorder by Ctrl+C I see console, but once I try any comand (''ls") it reports connection is broken. Any try to connect ssh again fails, but device responds to pings.

When I do the same using TTY (usb debug), than after programm finishes without any message about fail, I see login message. Once I try login with I see login message again and again.

### Steps to reproduce the problem:

 1. Start - stop adc_recoder few times using start script

pavel-demin, 2024-10-06:
Your problem description is too incomplete for me to understand exactly what you are doing, what startup script you are using, and why you are trying to interrupt the `adc_recorder` program while it is running. So if you want me to help you solve this problem, please provide more details (exact version of SD card image, scripts, commands, screenshots, ...). The more, the better.

Normally, the `adc_recorder` program runs for only a few seconds and does not need to be interrupted. If it runs longer, then it is likely that you did not configure the FPGA with the correct bitstream before running the `adc_recorder` program.

The `adc_recoder` is a very simple project and it is normal that it does not have the advanced features that you expect from it. For example, it does not have any signal handling. If you need this functionality, you will have to add it yourself.

mrengineer, 2024-10-06:
Dear Pavel, thank you for quick responce. I will provide all information you need, but little bit later

mrengineer, 2024-10-07:
Dear Pavel
Today I took releases/tag/20240204
1. I built the original adc_recorder and firmware for the FPGA.
Launched. Interrupted in the middle of the output. The ssh connection is normal, there is a response to the ls command.

2. Deleted the FIR.
The result is the same - everything is fine.

3. Deleted the CIC
The result is the same.

Printscreens are attached [printscreens.zip](https://github.com/user-attachments/files/17284373/printscreens.zip)

I come to the conclusion that my attempt to change the bus width of AXI to 128 bits goes from the ADC module to the packetizer led to the system crashing.

I will continue my step-by-step experiments and report to you, but later. Probably, at this point I will need your help.

_In general, I need to detect peaks in the sum of signals from two channels. But the signal is very noisy and has many peaks that should be discarded using mathematics and logic.

At the first step I need to calculate the sum of signals, the exponential average of this signal and the derivative. To debug I need to get a buffer of this data of 0.5 seconds at a speed of 125 MSPS

In the second step, I will debug the trigger autotuning algorithm, which will work either on the PS or on the PL. That's why a wide bus is needed - to give the count number, value and derivative to the PS

In the third step, the issue of false positives will be resolved_

pavel-demin, 2024-10-08:
Thank you for the details.

The throughput of the FPGA to RAM interface in the current version of the `adc_recorder` project is not sufficient to transfer ADC samples at 125 MSPS.

You will need to use the `HP0` interface and a different method to allocate the memory.

mrengineer, 2024-10-10:
Dear Pavel, thank you for your advice about HP0 and mememory

I've found https://github.com/Finnetrib/DmaTransfer/tree/master as example

The questions are
1. Maybe you know better example to start with? Please, give a link
2. Are necessary libraries will be in Alpine build releases/tag/20240204?
to use calls Xil_Out32(...) and 
#include "platform.h"
#include "xil_printf.h"
#include <xil_io.h>
#include "xil_cache.h"
#include "xil_mem.h"
and etc?

pavel-demin, 2024-10-10:
I think that the best examples are the ones that you write yourself based on the documentation provided by AMD/Xilinx.

The "Xil_" libraries and functions are part of the Standalone Board Support Package (BSP) and cannot be used from Linux userspace.

I am closing this issue because I do not think that using it to ask questions that are not directly related to the projects in this repository is the right approach.

closed 2024-10-10 by pavel-demin