pavel-demin/red-pitaya-notes #1157: Phase coherent inputs for GNU radio

mgarrett1955, 2024-10-24:
### Description of the setup:

 * Device:
 * SD card image:
 * Application:
 * Other relevant information:

### Description of the problem:

### Steps to reproduce the problem:

 1.
 2.
 3.

mgarrett1955, 2024-10-25:
Looks like from a previous questions that the VNA example has two synchronous recievers, I will review that and see if it can be adapted.

pavel-demin, 2024-10-26:
Although there are some projects  ([SDR transceiver compatible with HPSDR](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver-hpsdr/), [Vector Network Analyzer](http://pavel-demin.github.io/red-pitaya-notes/vna/)) in this repository that implement phase-coherent receivers, I do not think that their phase-coherent receivers can be easily used in GNU Radio.

Adding this functionality to the [SDR transceiver](http://pavel-demin.github.io/red-pitaya-notes/sdr-transceiver/) project would require an almost complete rewrite of that project.

I can only suggest trying to modify the [Wideband SDR receiver](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/sdr_receiver_wide/) project. I think that its receivers can be configured as two phase-coherent receivers relatively easily.