pavel-demin/red-pitaya-notes #1158: Unable to makefile due to scripts/core.tcl

minh5052vn, 2024-11-29:
Hi! I have some troubles with the Makefile where I filled out all the required information but it said there were some issues related to the core.tcl.
I wondered if I did it incorrectly or something else.
I hope you can help me! Thank you!
 
[image]
[image]

closed 2024-11-29 by pavel-demin

minh5052vn, 2024-12-02:
> Normally, it is not necessary to add `LD_LIBRARY_PATH` to `Makefile`. The Vitis/Vivado environment should be set by sourcing the `settings64.sh` script before running make.
> 
> The error message in the screenshot says that the `vivado` command was not found. I would say that you did not source the `settings64.sh` script before running make.
> 
> Most of my notes contain a "getting started" or "building from source" section that shows a sequence of commands to run. Here are some links to examples of these sections:
> 
> http://pavel-demin.github.io/red-pitaya-notes/led-blinker/#getting-started http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/#building-from-source

Hi Mr. Demin,
I wonder if the Makefile can work on Windows OS because on the my Linux server, the admin seems to hide or lock the pathway to Vivado folder where you can source the settings64.sh like your instruction.
Thank you for your help.

pavel-demin, 2024-12-02:
> I wonder if the Makefile can work on Windows OS

I do not know. I do not use Windows OS.

If you are using Windows OS, then it is possible to use Windows Subsystem for Linux (WSL) and install Ubuntu Linux via the application store. This will provide you with a Linux based environment that could be used to run AMD/Xilinx tools and my scripts.