pavel-demin/red-pitaya-notes #1160: Regarding TRX-DUO i2c Pin Assignment

bahari, 2024-12-07:


bahari, 2024-12-07:
Recently I'm in the middle to develop my own HF transceiver. During my low pass filter integration test via Quisk SDR application, I'm discover that TRX-DUO i2c pin (SCL & SDA) was opposite from what in the pin assignment documentation. From documentation it is clearly stated that pin 9 for SCL and pin 10 for SDA. When I wiring it up according to the documented pin assignment, my LPF selection was not functioning at all. I ran a debug print from Quisk SDR application, and the software clearly send the i2c logic command to TRX-DUO via metis protocol. When I twisted the i2c pin assignment, pin 9 to SDA and pin 10 to SCL, suddenly the i2c integration between TRX-DUO and PCA9555 working perfectly. Do any one discover this issues?

The pin assignment page:

https://elekitsorparts.com/trx-duo-sdr-on-board-connectors-and-ports-guide/

pavel-demin, 2024-12-11:
I think that you will have a better chance of getting an answer if you post this question to the NextGenSDRs mailing list:

https://groups.io/g/NextGenSDRs

Since I do not have TRX-DUO, I have no way to reproduce this problem.

closed 2024-12-11 by pavel-demin

bahari, 2024-12-12:
Very thanks for your info.