pavel-demin/red-pitaya-notes #1162: Decoding Problem with FT8 Transceiver

df6dbf, 2024-12-25:
Hello Pavel,
i have Problems to get the FT8 Skimmer working. Device is a Red Pitaya Stemlab 125-14 using the latest Image from your Site.
Call and Grid-Locator is added in upload-ft8.sh
The Bands are defined in write-c2-files.cfg.

In /dev/shm/decode-ft8.log I got the following Message.

Error termination. Backtrace:
Wed Dec 25 11:39:01 UTC 2024
Sleeping ...
Wed Dec 25 11:40:00 UTC 2024
Recording ...
Error on line 16 in configuration file.
Decoding ...
At line 33 of file ft8d.f90 (unit = 10)
Fortran runtime error: Cannot open file 'ft8_*_241225_1140.c2': No such file or directory

Resulting in no decodes in the decodes*.txt files and no Upload to PSKreporter.

73s Holger, DF6DBF

pavel-demin, 2024-12-25:
Hello Holger,

Have you edited the configuration file `write-c2-files.cfg`?

The error message indicates an error in this configuration file. Please check if all bands are separated by commas and that there is no trailing comma after the last band.

Best regards,

Pavel

df6dbf, 2024-12-25:
That is the point that confuses me:
There are no trailing comma after the last selected Band. :/

[image]

pavel-demin, 2024-12-25:
Thank you for the screenshot. There should be a comma on the uncommented line with the 21 MHz frequency.

df6dbf, 2024-12-25:
Oh! It seems i need new glasses soon! 
Thanks for your help and the excellent work!

Holger

closed 2024-12-25 by df6dbf