pavel-demin/red-pitaya-notes #1163: Request: Command Line INI/CFG Options for multiple TCP_SMEM instances

azwirko, 2024-12-28:
### Description of the setup:

 * Device: STEMLAB 125-14 (two or more simultaneously)
 * SD card image: 20241222
 * Application: sdr_receiver
 * Other relevant information: smem_tcp with intf_smem and SkimSrv integration

### Description of the problem/request:
I currently run CWSL_Tee https://github.com/HrochL/CWSL with two instances of SkimSrv where each one has its own install directory (SkimSrv, SkimSrv2) and each directory has its own copies of CWSL_Tee.dll/CWSL_Tee2.dll and CWSL_Tee.cfg/CWSL_Tee2.cfg file that allow pointing to two different Redpitaya MAC addresses.  A Windows batch file starts the entire system running from the command line including Aggregator without any other user interaction.  

First issue is TCP_SMEM seems to require a GUI interaction (Connect button press). Second it not does seem to allow for running two instances with two different IP addresses.  I copied the EXE and DLL files each to unique folders and names. I start the EXEs and enter different IP addresses.  I start two different instances of SkimSrv but the last SkimSrv app seems to cause the frequencies in both TCP_SMEM UIs to be the same.  In the SkimSrv apps the drop down only shows a single SMEM 0 while in CWSL_Tee the dropdown options names are unique.  

Is it possible to have an INI or CFG file added to TCP_SMEM to provide for the configuration of two or more instances and to start without GUI/user interaction and such that each has its own memory space addressable by two different SkimSrv instances?  

Thansk & 73!

andyz - K1RA

Single instance of TCP_SMEM (showing second directory for SkimSrv2 and second TCP_SMEM, not yet running)
[image]

[image]

Trying to run two instances of TCP_SMEM where first instance is overridden by values of second running SkimSrv instance
[image]

pavel-demin, 2024-12-28:
Hello Andy,

Thank you for testing the SDR receiver application. This is still a relatively new application that is used by very few people and I appreciate all the feedback.

It is already possible to use multiple SMEM instances. There is an example showing how this can be achieved at [this link](http://pavel-demin.github.io/red-pitaya-notes/sdr-receiver-122-88/#running-cw-skimmer-server-and-reverse-beacon-network-aggregator). By default, the SMEM index is 0. To use another index, it must be appended to the file names of programs and libraries. For example, `tcp_smem_1.exe`, `intf_smem_1.dll`, etc.

I will check later if it is possible to start the connection automatically when the `tcp_smem.exe` program starts. Hopefully it is as simple as calling `Connect` right after `Init`. Otherwise it would be easier for me to add a non-GUI program with all parameters passed via command line arguments.

All settings are stored in the registry and at the moment I would prefer not to have configuration files.

Best regards,

Pavel

azwirko, 2024-12-29:
Thanks Pavel - I'd been following these instructions https://pavel-demin.github.io/red-pitaya-notes/sdr-receiver/ as I didn't have an 122-16, but I tried them and it works as expected with two 125-14s.  I'm now trying to understand the cat_smem, file_smem and wave_smem apps you've written to see how they might be used to emulate something similar to CWSL_Digi https://github.com/alexranaldi/CWSL_DIGI that I'd used in conjunction with CWSL_Tee.  Your SMEM system seems to have fewer requirements than CWSL and is seems more streamlined.  Thanks for your efforts!

andyz - K1RA

pavel-demin, 2024-12-29:
I think that it would be quite difficult to have CWSL_DIGI functionality with the current state of this project.

I can think of the following approach to reimplement the CWSL_DIGI functionality:
- add `dump_smem.exe` that would write I/Q samples to a file
- write a program in Python that would
  - periodically run `dump_smem.exe`
  - preprocess I/Q samples using `numpy`
  - call the decoders (`wsprd` and `jt9`)
  - upload results to wsprnet and pskreporter

azwirko, 2024-12-29:
Thanks for the ideas.  Can you provide any high level help or example usage for your file_smem, cat_smem and wave_smem apps?  I was wondering if any of those could be leveraged with the WSJT-X jt9 command line decoder.  I was thinking if any of your apps produced WAV files that might be another means to perform the decoding with a Python wrapper and then parse the text decode output and upload to PSKr and WSPRnet which I've done in the past.

pavel-demin, 2024-12-29:
> Can you provide any high level help or example usage for your file_smem, cat_smem and wave_smem apps?

`wave_smem.exe` is used to send I/Q samples from one of the SMEM I/Q channels to CW Skimmer via a virtual audio device and `cat_smem.exe` is used to control CW Skimmer from Thetis. These two programs can be used as shown in [this video](https://www.youtube.com/watch?v=t9WU99gfP_0).

`file_smem.exe` reads I/Q samples from a file and sends them to a virtual audio device and a SMEM I/Q channel. I used it to simultaneously test the dynamic ranges of CW Skimmer and CW Skimmer Server.

>  I was wondering if any of those could be leveraged with the WSJT-X jt9 command line decoder.

Implementing the CWSL_DIGI functionality was not the goal of this project when I worked on it last year.

I could add `dump_smem.exe` and Python code to convert the I/Q samples to WAV files for all the sub-bands shown in the table at [this link](https://groups.io/g/NextGenSDRs/message/2784).

azwirko, 2024-12-29:
Appreciate the explanation. I didn't expect CWSL_Digi functionality, but was curious to what extent any of your other tools provided mirrored the CWSL_Wave, CWSL_File, CWSL_Net, capabilities.  

So now that I have a working SMEM, I guess the only request is to auto connect on startup if possible as I understand configuration is stored in the registry.  That would allow me to script starting the system without any other user interaction.

Thanks again & 73.

andyz - K1RA

pavel-demin, 2024-12-29:
Calling `Connect` right after `Init` seems to work and starts the connection automatically when the `tcp_smem.exe` program starts.

Here is a link to the pre-built SMEM binaries with this modification:

https://www.dropbox.com/scl/fi/kto4dgsn1dh58saohc5r6/sdr-smem-20241229.zip?rlkey=8effibpq4r582wepjee7cs8d0&dl=1

azwirko, 2024-12-29:
Thanks for the quick turn around. Works great.  Happy New Year.