pavel-demin/red-pitaya-notes #1164: Bootable SD from http://pavel-demin.github.io/red-pitaya-notes/alpine/ = > not working

Kervenou, 2025-01-10:
Hello, 
I don't pass the alpine installation. I'm new to this system, so I apologize for this naive question.

### Description of the setup:
 * Device: SDRlab 122-16 V1.1 connected to a PC Windows10 via a router
 * SD card image: red-pitaya-alpine-3.20-armv7-20241222.zip
 * Application: SDR 
 
I followed the instructions for "Alpine with pre-built applications", inserted the SD card in the Redpitaya.
The green LED is on, with the USB/serial console, the boot process is beginning nicely.
But at the Mounting boot media step, the following message appears  :

----
 * Mounting boot media: /dev/mmcblk0p1: Can't lookup blockdev
/dev/mmcblk0p1: Can't lookup blockdev
/dev/mmcblk0p1: Can't lookup blockdev
/dev/mmcblk0p1: Can't lookup blockdev
mount: mounting /dev/mmcblk0p1 on /media/mmcblk0p1 failed: No such file or directory
ls: /media/mmcblk0p1/*Mounting boot media: failed.
.apkovl.tar.gz: No such file or directory failed.
initramfs emergency recovery shell launched. Type 'exit' to continue boot
----

Have you a clue ?

Sincerely yours,

Hervé

pavel-demin, 2025-01-11:
I have just downloaded the zip file and tested it with SDRlab 122-16. The system booted without any problems. So maybe the problem is with your SD card. Please make sure to use an SD or SDHC card with a capacity from 1GB to 32GB.

Kervenou, 2025-01-12:
Hello,

Thank you for your reply.

I made new tests : I tried with two different SD cards (4 and 16 GB) and 
different versions (20240204 and 20241222).
I have the same problem.
I'm attaching the console dump. I see the first possible problem on line 19:
Kernel command line: console=ttyPS0,115200 earlycon earlyprintk 
initrd=0x3000000,4M modloop=modloop cma=36M
Unknown kernel command line parameters "earlyprintk modloop=modloop", 
will be passed to user space
and then in 205 :
  * Mounting boot media: /dev/mmcblk0p1: Can't lookup blockdev

These SD cards have been formatted under Windows 10 in FAT32: maybe 
that's the problem?

I successfully flashed a new version of Reptipaya version 2.00 37
but it was from an image with balenaEtcher.

Thanks a lot in advance for your help !
Sincerely yours,

Hervé

Booting Linux on physical CPU 0x0
Linux version 6.6.32-xilinx ***@***.***) (arm-xilinx-linux-gnueabi-gcc.real (GCC) 13.3.0, GNU ld (GNU Binutils) 2.42.0.20240216) #1 SMP PREEMPT Sat Dec 21 17:32:03 CET 2024
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: xlnx,zynq-7000
earlycon: cdns0 at MMIO 0xe0000000 (options '115200n8')
printk: bootconsole [cdns0] enabled
Memory policy: Data cache writealloc
cma: Reserved 36 MiB at 0x1dc00000 on node -1
Zone ranges:
  Normal   [mem 0x0000000000000000-0x000000001fffffff]
  HighMem  empty
Movable zone start for each node
Early memory node ranges
  node   0: [mem 0x0000000000000000-0x000000001fffffff]
Initmem setup node 0 [mem 0x0000000000000000-0x000000001fffffff]
percpu: Embedded 13 pages/cpu s23188 r8192 d21868 u53248
Kernel command line: console=ttyPS0,115200 earlycon earlyprintk initrd=0x3000000,4M modloop=modloop cma=36M
Unknown kernel command line parameters "earlyprintk modloop=modloop", will be passed to user space.
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Built 1 zonelists, mobility grouping on.  Total pages: 130048
mem auto-init: stack:all(zero), heap alloc:off, heap free:off
Memory: 464344K/524288K available (9216K kernel code, 257K rwdata, 2148K rodata, 1024K init, 155K bss, 23080K reserved, 36864K cma-reserved, 0K highmem)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
rcu: Preemptible hierarchical RCU implementation.
rcu:    RCU event tracing is enabled.
rcu:    RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
        Trampoline variant of Tasks RCU enabled.
        Tracing variant of Tasks RCU enabled.
rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
slcr mapped to (ptrval)
GIC physical location is 0xf8f01000
L2C: platform modifies aux control register: 0x72360000 -> 0x72760000
L2C: DT/platform modifies aux control register: 0x72360000 -> 0x72760000
L2C-310 erratum 769419 enabled
L2C-310 enabling early BRESP for Cortex-A9
L2C-310 full line of zeros enabled for Cortex-A9
L2C-310 ID prefetch enabled, offset 1 lines
L2C-310 dynamic clock gating enabled, standby mode enabled
L2C-310 cache controller enabled, 8 ways, 512 kB
L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76760001
[... 146 lines cut ...]
mmcblk0: mmc0:0001 SD 15.0 GiB
 mmcblk0:
Loading compiled-in X.509 certificates
clk: Disabling unused clocks
ALSA device list:
  No soundcards found.
Freeing unused kernel image (initmem) memory: 1024K
Run /init as init process
Alpine Init 3.10.1-r0
Alpine Init 3.10.1-r0
Loading boot drivers...
 * Loading boot drivers: Loading boot drivers: ok.
ok.
Mounting boot media...
 * Mounting boot media: /dev/mmcblk0p1: Can't lookup blockdev
/dev/mmcblk0p1: Can't lookup blockdev
/dev/mmcblk0p1: Can't lookup blockdev
/dev/mmcblk0p1: Can't lookup blockdev
mount: mounting /dev/mmcblk0p1 on /media/mmcblk0p1 failed: No such file or directory
ls: /media/mmcblk0p1/Mounting boot media: failed.
*.apkovl.tar.gz: No such file or directory
failed.
initramfs emergency recovery shell launched. Type 'exit' to continue boot
sh: can't access tty; job control turned off
~ #

pavel-demin, 2025-01-12:
The following lines in your log show that Linux does not see the partitions on the SD card:
```
mmc0: new high speed SDHC card at address 0001
mmcblk0: mmc0:0001 SD 15.0 GiB
 mmcblk0:
```

No partition is shown on the last line.

This is what these lines look like on my system:
```
mmc0: new high speed SDHC card at address 0007
mmcblk0: mmc0:0007 SD32G 29.0 GiB
 mmcblk0: p1
```

The partition `p1` is shown on the last line.

At the moment I have no idea why Linux does not see the partitions on your SD card and how to fix this problem.

dc7ds, 2025-01-15:
Or try using https://www.sdcard.org/downloads/formatter/BRStefan

Have you tried to write an empty SD card image with a single FAT32 partition from this link? This way you will have a properly formatted SD card without using Windows 10 formatting tools.

StefanDausP, 2025-01-15:
SD Memory Card Formatter for Windows/Mac | SD Associationsdcard.org

Or try using https://www.sdcard.org/downloads/formatter/BRStefan

pavel-demin, 2025-01-15:
I have just found very clear instructions on how to repartition and reformat an SD card using the tools that come with Windows:

https://www.radishlogic.com/raspberry-pi/how-to-format-an-sd-card-after-using-it-on-a-raspberry-pi-windows-disk-management/

If I had a computer with Windows, I would try to follow these instructions.

Kervenou, 2025-01-15:
Hello,

I followed the link provided by Stefan and downloaded this SD card 
formatter for Windows :
https://www.sdcard.org/downloads/formatter/sd-memory-card-formatter-for-windows-download/ 

After the format, I then copied your files, and this time it works !
Bu the way I checked the SD card with the windows disk manager, before 
and after the SD card formatter : no difference, but for sure there is one !

I would I would like to thank you for your help.
Now begin the hard work for me to connect the Redpitaya with a Windows 
SDR program ...

Sincerely yours,

Hervé

-- 
-----------------------------------------------
Hervé Saint-Jalmes

LTSI, INSERM U1099

Bâtiment 5, PRISM Villejean
Faculté de Médecine
Université Rennes 1

CS 34317
35043 Rennes Cedex
France
02 23 23 48 49

----------------------------------------------

closed 2025-01-15 by pavel-demin