pavel-demin/red-pitaya-notes #1165: wave_smem crashes

RogerD73, 2025-01-25:
Hi

RogerD73, 2025-01-25:
Hi
I'm using a QMTECH XC7Z020.

I'm having an issue with wave_smem.  I can run tcp_smem and extio_smem with HDSDR OK.  I can run tcp_smem with intf_smem with skimmer server OK.

However, if I try to run wave_smem.exe it crashes.  I've managed to see the "error message".  The program opens, displays the available audio devices and then crashes.  I've tried this on PCs running Windows 7, 10 and 11 with the same result. 

I'm sure that this is operator error, but I would really appreciate help to get this running so I can pass the IQ sream to another program.      
Thanks in advance.
Roger G4BVY

pavel-demin, 2025-01-25:
Thank you for testing the SDR receiver application and SDR SMEM programs.

If you run `wave_smem.exe` without arguments, then it exists showing available audio devices.

I think that you have to run it from the command line. Once without arguments to see the available audio devices. Then with two arguments (`wave_smem.exe audio_device_number smem_channel_number`). The command with arguments can be saved in a batch file or in a desktop shortcut.

RogerD73, 2025-01-25:
Hi Pavel

RogerD73, 2025-01-25:
Hi Pavel
Many thanks for the prompt response.  I've now got "something" going down a virtual audio channel from tcp_smem through wave_smem to my external application.  I need to experiment a bit as I think I have a mismatch of data rates somewhere.  But this is great progress!!

I now need to get the rf frequencies synchronised.  I'm guessing that cat_smem.exe may be the answer?  I've run that from a command prompt and get that the usage is: port chan chan call port port
Do you have the definitions of what these refer to?

Thanks for your help
Roger

pavel-demin, 2025-01-25:
`cat_smem.exe` is used to read frequencies from Thetis via a serial port and send them to two CW Skimmers via telnet. It also sets the frequencies of the two SMEM channels used by the CW Skimmers. The arguments are:
- serial port number
- SMEM channel for the fist CW Skimmer
- SMEM channel for the second CW Skimmer
- call sign
- telnet port for the fist CW Skimmer
- telnet port for the second CW Skimmer

If you are not using Thetis and CW Skimmer, then I think that you will need to use a different program to synchronize the frequencies.

closed 2025-02-05 by pavel-demin