pavel-demin/red-pitaya-notes #1166: MCPHA pha-start.c issue

PeteiThemba, 2025-02-03:
Hi,
 I'm using the latest release (red-pitaya-notes-20241222) and although the python mcpha.py works, I cannot collect data from the command line through pha-start:

pavel-demin, 2025-02-04:
Thank you for testing `pha-server` and finding this problem.

I have just fixed this problem with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/93ef583c73b1b25618cc681f6a9ca43904bec879).

PeteiThemba, 2025-02-05:
Thanks, looks good.  Although the data compared to a previous release seems to be "reversed"; from hexdump:

Previous version:
0000000 3cdb 0004 0000 0000 2b66 0000 0000 0000
0000010 4fdf 0005 0000 0000 225e 0000 0000 0000
0000020 c147 0006 0000 0000 217c 0000 0000 0000
0000030 8ed3 000a 0000 0000 1de9 0000 0000 0000
0000040 d5cb 000c 0000 0000 2a75 0000 0000 0000

(timestamp, then two channel data, channel 1 in use)

This version:
0000000 0000 0000 0633 0000 0000 0000 0459 0000
0000010 0000 0000 0078 0000 0000 0000 09bd 0000
0000020 0000 0000 008a 0000 0000 0000 6a8d 0001
0000030 0000 0000 084e 0000 0000 0000 1da9 0002
0000040 0000 0000 0092 0000 0000 0000 2341 0002

(two channel data, channel 1 in use then timstamp)

I can modify my code to recognise the data format but I'm curious to know why it looks like this :)

pavel-demin, 2025-02-05:
Thank you for testing the new version of the code.

I originally used a FIFO buffer that did not order the data and timestamps correctly and I had to add a workaround. I recently switched to a FIFO buffer that works better but I forgot to remove the workaround from the MCPHA application. This workaround is now removed with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/f17132fff4c5c4ed045403f7bbbd47c2feb6a73e).

Here is a link to a new bitstream file that contains this fix:

https://www.dropbox.com/scl/fi/q94eyft6inlq37cclldy6/mcpha.bit?rlkey=mx47csrz4k8a6zqxbwn5dbgsd&dl=1

PeteiThemba, 2025-02-06:
Looks good, thanks for the update !

Something I notice running two independant channels is that it seems the value of the channel is not zeroed before/after being put in the buffer.  I can interpret below that 0x15a0, 0x15b0 is channel 1 update, but channel 2 has the previous value.  0x15e0 has the new value for channel 2, but the channel 1 has it's previous value.

(On a brief test this afternoon I found that some values could be output below the threshold applies, for example a value of 0x0001 but I could not replicate this issue consistently - but I thought I may mention it at this point)

hexdump snippet:

00015a0 8d55 00af 0000 0000 00d3 0000 0430 0000
00015b0 ac31 00af 0000 0000 0079 0000 0430 0000
00015c0 0029 00b0 0000 0000 0376 0000 0430 0000
00015d0 04cd 00b0 0000 0000 002b 0000 0430 0000
00015e0 4d71 00b1 0000 0000 002b 0000 042e 0000
00015f0 505d 00b1 0000 0000 0092 0000 042e 0000
0001600 8179 00b1 0000 0000 0086 0000 042e 0000
0001610 8fcd 00b1 0000 0000 008a 0000 042e 0000
0001620 b0a5 00b1 0000 0000 02e7 0000 042e 0000
0001630 7a4d 00b2 0000 0000 02c1 0000 042e 0000
0001640 7f05 00b2 0000 0000 0031 0000 042e 0000
0001650 35b9 00b3 0000 0000 0031 0000 0430 0000
0001660 6cc9 00b3 0000 0000 023a 0000 0430 0000
0001670 26cd 00b4 0000 0000 02fa 0000 0430 0000
0001680 2b2d 00b4 0000 0000 0027 0000 0430 0000
0001690 0489 00b5 0000 0000 0084 0000 0430 0000
00016a0 1e01 00b5 0000 0000 0084 0000 042f 0000
00016b0 0649 00b7 0000 0000 0084 0000 042e 0000
00016c0 dda1 00b7 0000 0000 087e 0000 042e 0000
00016d0 e2fd 00b7 0000 0000 009f 0000 042e 0000
00016e0 f0ad 00b7 0000 0000 007e 0000 042e 0000
00016f0 ee91 00b8 0000 0000 007e 0000 042c 0000

pavel-demin, 2025-02-06:
Thank you for testing the new version and finding another problem.

It seems that I removed with [this commit](https://github.com/pavel-demin/red-pitaya-notes/commit/8841335f70678bc4e354d22918dad59ea154fc0e) the part of the [axis_pulse_height_analyzer](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_pulse_height_analyzer.v) module that sets the data to zero when there is no valid data. I have restored this part with [this new commit](https://github.com/pavel-demin/red-pitaya-notes/commit/f98d4614154f789dc0784be25a3e5fa0ce982f10).

Here is a link to an updated bitstream file:

https://www.dropbox.com/scl/fi/q94eyft6inlq37cclldy6/mcpha.bit?rlkey=mx47csrz4k8a6zqxbwn5dbgsd&dl=1

PeteiThemba, 2025-02-07:
Looks good !  Thanks for the fix !