pavel-demin/red-pitaya-notes #1167: The DAC is taking the double amount of clocks to output

tandeitnik, 2025-02-04:
Hello,

I have a Vivado project that takes 32 clock cycles to output two 14-bit results. I'm wiring both results so that one is outputted in Red Pitaya's Output 1 and the other in Output 2. Using an oscilloscope, I noticed that each output is updated every 64 clock cycles, precisely double the expected number of cycles. Can you explain why this is happening and if there is a way to correct it?

Thanks,

Daniel.

pavel-demin, 2025-02-04:
I am afraid that the information that you provide is not enough for me to understand what you are doing. If you are using the [axis_red_pitaya_dac](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_red_pitaya_dac.v) module from this repository, then please provide a description or diagram of how this module is used in your project. A screenshot of what the oscilloscope shows and a diagram of what you expect might also help me understand the problem.

At the moment the only possible source of the problem that I can think of is that the [axis_red_pitaya_dac](https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_red_pitaya_dac.v) module is probably not connected to the right clock signals.

tandeitnik, 2025-02-04:
Hi,

sorry for the lack of detail. Here is an image of how the output of the red pitaya is generated

[image]

As I mentioned, the output_A and output_B pins receive 14-bit signals that should change every 32 clock cycles. Here is an image of my oscilloscope where I measured the time difference between two sequential outputs. Doing the math, you conclude that the output is changing every 64 clock cycles:

[image]

The aclk pin of the AXI4-Stream Red Pitaya DAC is connected to the adc_clk of the AXI4-Strem Red Pitaya ADC. Please tell me if you need more information.

pavel-demin, 2025-02-04:
Thank you for the additional details.

What is the output frequency and phase of output `clk_wiz_1`?

It seems that you are using a very old version of my code. I would recommend you to use the latest version of the code from this repository and use the [template](https://github.com/pavel-demin/red-pitaya-notes/tree/master/projects/template) project as a starting point.

tandeitnik, 2025-02-05:
I'm assuming you are asking for the `clk_wiz_0` since there is no `clk_wiz_1`.  These are its current configuration:

[image]

[image]

Thanks for sharing an updated version of your code. I will try to update mine to it.

tandeitnik, 2025-02-05:
Hi,

I'm unsure how to use the `block_design.tcl` file as a template. I tried to use the 'Run TCL script...' tool from Vivado, but it returned an error. I got all my projects running based on the projects provided by Anton Potočnik in his blog.

pavel-demin, 2025-02-05:
The output frequency is OK but I am not sure about the phase. I remember setting the phase to `-90.0` when I used a similar setup.

I still do not see any major problem with the part of your project shown in the screenshot. So I am starting to think that the source of the problem is probably in the part of your project that is not shown in the screenshot.

The version of my code that Anton Potočnik used in his tutorials many years ago is now very old. He also used a different approach to working with Vivado.

Instructions on how to build projects from this repository can be found in my notes at [this link](http://pavel-demin.github.io/red-pitaya-notes/led-blinker/#getting-started).

closed 2025-02-05 by pavel-demin