pavel-demin/red-pitaya-notes #1170: Not serving interface wlan0

thxred, 2025-03-25:
Could it be my configuration, or is red-pitaya-notes not serving on the wlan0 interface?

My wlan0 interface seems to be configured correctly, however only the eth0 interface works.

[image]

Thank you.

pavel-demin, 2025-03-25:
Please provide the following information that would help me better understand your issue:
- device
- SD card image
- steps to reproduce the problem
- other relevant information

thxred, 2025-03-25:
I should have mentioned the necessary information, my apologies. 

The device I am using is a WEB-888SDR with the latest image: red-pitaya-alpine-3.20-armv7-20241228.

I used the steps to configure the WIFI (client, not hotspot) through the official Alpine Linux page. 

`wpa_passphrase 'ExampleWifiSSID' 'ExampleWifiPassword' > /etc/wpa_supplicant/wpa_supplicant.conf`

Because I couldn't get it to work with the script /root/wifi/client.sh provided in the image.

Then I used `lbu commit -d` to save the configuration permanent on the card. 

After rebooting I can access via ssh to both interfaces, but access the red-pitaya-notes http page only through the eth0 interface.

eth0
[image]

wlan0
[image]

pavel-demin, 2025-03-25:
Thank you for this additional information.

It looks like you are not using an SD card image from this repository. Release 20241228 does not exist in this repository.

I think that the SD card image that you are using is from the following repository:

https://github.com/RaspSDR/red-pitaya-notes

If SSH works with both interfaces, then I do not really see why the web server is not accessible via the wireless interface.

I think that it is quite important to run the `wifi/client.sh` script to properly configure network related services. Do you see any error messages when you run this script?

The wireless connection is much slower than the wired connection. It is only useful with the last two applications in your screenshot (WSPR and FT8 receivers), as they do not transfer much data over the network. So maybe you do not really need the wireless connection.

pavel-demin, 2025-03-25:
I have just realized that the `eth0` and `wlan0` interfaces are on the same subnet. Linux is not configured by default to work properly with this type of configuration:

https://access.redhat.com/solutions/30564

If I remember correctly, the following commands can be used to solve this problem:
```
echo 2 > /proc/sys/net/ipv4/conf/all/arp_announce
echo 1 > /proc/sys/net/ipv4/conf/all/arp_ignore
echo 2 > /proc/sys/net/ipv4/conf/all/rp_filter
```

thxred, 2025-03-26:
Since only one interface is "physically connected", one at a time, I think it shouldn't be a problem the same subnet configuration, also have no intention of using both at the same time.

The bad news is really the WIFI bandwidth, but we can't have the best of both worlds, I think.

This is the setup I'm trying to build:
```
+--------------------+ 
|      Web888SDR     |
|  Red-Pitaya-Notes  |
+---------+----------+
         WIFI    
+---------+----------+
|      Switch/AP     |
+---------+----------+
       ETHERNET
+---------+----------+
|        RPI         |
|     OpenWebRX++    |
+--------------------+
```
I'll try using the commands you provided and see if anything changes.

Thank you

closed 2025-04-02 by thxred