pavel-demin/red-pitaya-notes #1172: DAC & DMA noise question

Laswer5, 2025-04-01:
**Description of the setup:**
* Device: STEMlab 125-14
* SD Card image: Red Pitaya's v2.05-37
* Cores in question: axis_red_pitaya_dac, axis_ram_reader (also using axis_ram_writer and axis_red_pitaya_adc)
* Other relevant information: cores have been modified, detailed below

## Description of problem / request

First off, I'd like to thank you for all the information and cores that you have created about the Red Pitaya, these resources have been indispensable for my understanding of the pitaya and programming its FPGA.  I am a complete novice at FPGA programming, so I apologize if I've overlooked something trivial or if this question is outside the scope of this issue board. 

I have the following situation; I need to generate arbitrary waveforms of quite long periods (milliseconds) at full sampling rate because of high frequency components. Additionally, we only capture one period at a time, so no continuous generation/acquisition. This won't work with the regular AWG API due to the small buffer of course, so I devised the following setup:

Write samples in the correct format to RAM -> Issue a trigger -> axis_ram_reader starts reading samples into the DAC and the ADC starts writing samples to another location in RAM. 

The ADC to ram pipeline seems to be working fine; I've compared the signal from my ADC and the regular acquisition with an external function generator, and I seem to get the same results (minus some minor filtering, but I don't apply any in hardware so that's to be expected). 

However, the signal from the DAC has some strange effects that I haven't been able to figure out. The signal seems to be distorted by some kind of periodic ringing pulses. At low frequencies these can be filtered away, but it gets problematic as we get into the MHz range where we have frequencies of interest. 

I would suspect bad wires (breadboard) but I don't get this kind of distortion with the regular red pitaya API, so I'm lead to the conclusion that I'm doing something wrong. 

If you have any insight / suggestions, this would be immensely helpful. I will provide some additional information and images of the signal effects below. 

## Modifications to the cores
I am only interested in using one channel, so I've modified the ADC and DAC to only use channel A. Additionally, I have reduced the tdata width to 16 bits in order to store the samples consecutively. 

For the DAC, I simply drive both edges with the same sample, since I don't really care about what happens on channel B. I don't know if this causes any issues. I've noted the phase for the clocks from your other projects, and I am using those values for the PLL clock. 
```verilog

    // Both DDR edges output the same channel A sample.
    genvar j;
    generate
        for (j = 0; j < DAC_DATA_WIDTH; j = j + 1)
        begin : DAC_DAT
            ODDR #(
                .DDR_CLK_EDGE("SAME_EDGE"),
                .INIT(1'b0)
            ) ODDR_inst (
                .Q(dac_dat[j]),
                .D1(int_dat_a_reg[j]),
                .D2(int_dat_a_reg[j]),
                .C(aclk),
                .CE(1'b1),
                .R(1'b0),
                .S(1'b0)
            );
        end
    endgenerate

    // Indicate that this core is always ready to accept new AXIS data.
    assign s_axis_tready = 1'b1;

endmodule
```

I instantiate the ram reader as follows to use the 16 bit width:

```verilog
RAM_RDR #(
    .ADDR_WIDTH(16),
    .AXI_ID_WIDTH(6),
    .AXI_ADDR_WIDTH(32),
    .AXI_DATA_WIDTH(64),
    .AXIS_TDATA_WIDTH(16),
    .FIFO_WRITE_DEPTH(512)
  )
```

# Some images of the received DAC signal

# 5KHz sine wave full and zoomed in on noise

[image]
[image]

## Sine low-pass filtered at 10KHz

[image]

# More complex waveform in the MHz range

[image]

[image]

pavel-demin, 2025-04-04:
Thank you for the detailed description of your project.

I do not see any problem with your code.

The ADC and DAC modules can be used without modifications. You can connect only the first 16 wires of their interfaces and keep the other 16 wires disconnected.

I have never tried reading and writing samples simultaneously at full sample rate, so I do not have any working examples of this kind of configuration.

I can think of several potential sources of the problem:
- some bugs in the reader and writer modules that I had not found before
- the reader and writer modules are connected to ports with insufficient bandwidth or non-optimally configured priorities
- insufficient size of FIFO buffers
- memory cache problems

Laswer5, 2025-04-04:
Thank you in any case for answering, I'll be sure to reply with what I've found out if I find a solution.

Laswer5, 2025-04-11:
I managed to solve it. I had used the connection automation in Vivado which connects the cores through interconnects. Connecting them directly to the HP ports worked.

closed 2025-04-11 by Laswer5