pavel-demin/red-pitaya-notes #1175: Sdr_transiever Application working with different boards

GoSamOfficial, 2025-04-26:
Hi Paverl! I hope you are doing well. I have a quick question. I use SdrLab 122-16. On the latest red pitaya OS, does sdr_transiever application decides by itself if it's steamlab or sdrlab connected? I made some tests using gnuradio modules. With default sdr_transiever gnuradio modules it doesn't work good, but with sdr_transiever_122_88 gnuradio modules works fine. But i am qurious that we are only changing some python libraries but not fpga, how does it work then? Thanks in advance!

pavel-demin, 2025-04-26:
I have never used SDR applications with recent versions of the SD card images provided by Red Pitaya, so I have no idea how exactly they work together.

Based on your description, I would say that the FPGA is configured with a compatible FPGA configuration. Otherwise it would not work at all.

The correct Python libraries are indeed those in the `sdr_transiever_122_88` project.

closed 2025-04-29 by pavel-demin

GoSamOfficial, 2025-05-01:
@pavel-demin but do you still have ADC/DAC ip's for vivado?

pavel-demin, 2025-05-01:
I am not sure if I understand the meaning of the last question correctly and how it relates to my previous answer.

To make sure the IP cores have not disappeared from the repository, I have just checked it.

It seems that the code for the ADC and DAC interface IP cores is still present in this repository. Here are the links to their source files:

https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_red_pitaya_adc.v

https://github.com/pavel-demin/red-pitaya-notes/blob/master/cores/axis_red_pitaya_dac.v

If the above does not answer your question, please provide me with more context.